basler/pypylon #672: Runtime Exception: Camera device physically removed (in instantcameraimpl.h at line 2079)

hsutang, 2023-10-27:
I'm encountering a runtime exception when working with the camera. The error message I receive is:

> The camera device has been physically removed runtime exception thrown (file ‘instantcameraimpl.h’, line 2079).

I've ensured that the camera is properly connected via USB 3.0 and the necessary drivers are installed. I've also tried [any other troubleshooting steps you've taken, like "restarting the computer", "trying a different cable", etc.]. 

Has anyone else faced this issue? Any suggestions or workarounds would be greatly appreciated. 

Thank you in advance for your assistance!

varshinimuthukumar1, 2023-10-27:
I have the same error.  I do perform a `DeviceReset()` (method available for the camera instance) in case of exception.  But this does not help.  
The only solution is to disconnect and reconnect the camera physically.  Could someone help with some pointers to fix this issue?
Or suggest some ideas to do some other kind of reset- so the camera starts getting recognized?

HighImp, 2023-10-30:
All commands to the camera do not help, because the exception says that the device is considered disconnected/removed. Is the camera still visible in Pylon Viewer after the error? If so, you should be able to enumerate the camera and create a new instance. Otherwise, as far as I know, there is no other way to create a new connection other than a powercycle.  A disconnect shouldn't happen, though. Is there any particular code you can provide to trigger the problem?