basler/pypylon #676: Camera unable to grab frames after PC resume from suspend (Fedora 38)

jokubas11, 2023-11-06:
I am using acA2040-90umNIR camera. I have run into the problem, where at first camera works well, if I walk away for some time and suspend the PC, once the user resumes from suspend, the camera is unable to grab frames anymore. The error message does not help very much: "Payload data has been discarded. Payload data can be discarded by the camera device if the available bandwidth is insufficient".

- Plugging in and out the USB cable solves the issue, but the cable is buried deep, so it is not convenient to do this everytime + multiple people are using my app, so if new person uses the app, they get confused why is the camera not working.
- If the camera was open when I physically plug in and out the device, the issue is not rectified then.
- Happens only on Fedora 38, never had it happen on Ubuntu or Windows. Switching to different OS is not an option for me.
- Happens with all USB cameras that I am using.
- Camera is still detectable and it is possible to control it, adjust settings and etc., it's just that it is not possible to acquire frames.
- It does not matter whether the camera was open or not when the PC was suspended, the issue persists regardless of whether it was open during suspension, once user returns from suspension, it is not possible to grab frames.
- At first I thought maybe in suspension mode USB port has its power disabled and is unable to power back up, but disabling usb autosuspend did not solve the issue
- Log from dmesg does not indicate any issue

I wonder if anyone had similar issues and had success solving it

SMA2016a, 2023-11-12:
first you can call camera.DeviceReset.Excicute()   if this does not helps.

run following command on linux console
sudo usb_modeswitch -v 0x2676 -p 0xba02 --reset-usb

if this also not help to bring the system in good shape, try to reset the USB hostcontrller.

try to reset the camera using Linux command

sudo usb_modeswitch -v 0x2676 -p 0xba02 -b 2 -g 1 --reset-usb

It aims to reset the specific device and not every ace camera.
The -b option chooses the bus while -g chooses the device ID number.   you need to adapt this value on your system.

lsusb will give such the needed information of bus ID

dhrosquete, 2024-02-22:
Hi @SMA2016a I have tried these commands in pypylon 1.8 and pylon 5.1, but the cameras still are not responding.

Do you know what is behind the courtains in the reset that pylon viewer has? Is there a command to emulate this on terminal or through the API in python?

dk2710, 2026-09-17:
Hello @jokubas11 
Did you found a fix for that to reset the camera in anyway?