basler/pypylon #677: Manually setting  white balance ratio.

Septembit, 2023-11-07:
Hello, do anyone know in pypylon there is way of setting white balance manually. Just like C++ does in this link [https://docs.baslerweb.com/balance-white](https://docs.baslerweb.com/balance-white).

```
// Set the red intensity to 108.789%
camera.BalanceRatioSelector.SetValue(BalanceRatioSelector_Red);
camera.BalanceRatio.SetValue(1.08789);
// Set the green intensity to 100%
camera.BalanceRatioSelector.SetValue(BalanceRatioSelector_Green);
camera.BalanceRatio.SetValue(1.0);
// Set the blue intensity to 219.678%
camera.BalanceRatioSelector.SetValue(BalanceRatioSelector_Blue);
camera.BalanceRatio.SetValue(2.19678);
```

thiesmoeller, 2023-11-07:
Hi @Septembit,

the c++ API and the pypylon API are very close to each other.
See [pypylon-samples](https://github.com/basler/pypylon-samples) for in depth background and documentation.

Typically a search'n'replace is enough ;-)

You can use this sed script to convert the C++ Basler product documentation code snippets to python
```bash
cat <sample_code.cpp> | sed -E 's/(\S+)\.(\S+).SetValue\(\s*\2_(\S+)\)/\1.\2\.SetValue("\3")/;s/^\/\//#/;s/SetValue\((.*)\)/Value = \1/;s/;\s*$//'
```

so your request gets:

```python
# Set the red intensity to 108.789%
camera.BalanceRatioSelector.Value = "Red"
camera.BalanceRatio.Value = 1.08789
# Set the green intensity to 100%
camera.BalanceRatioSelector.Value = "Green"
camera.BalanceRatio.Value = 1.0
# Set the blue intensity to 219.678%
camera.BalanceRatioSelector.Value = "Blue"
camera.BalanceRatio.Value = 2.19678
```

If you don't have `sed` on your system you can use an online tool like https://sed.js.org/ with `-E 's/(\S+)\.(\S+).SetValue\(\s*\2_(\S+)\)/\1.\2\.SetValue("\3")/;s/^\/\//#/;s/SetValue\((.*)\)/Value = \1/;s/;\s*$//'` as `Command line`

BatuhanVisys, 2024-05-08:
How can I change it in python ?

thiesmoeller, 2024-05-08:
You can look up every camera feature in our products documentation docs.baslerweb.com
We have python sample code for every feature online.

https://docs.baslerweb.com/balance-white#sample-code

BatuhanVisys, 2024-05-08:
I found the page. Thank you !

closed 2024-05-08 by thiesmoeller