basler/pypylon #679: Camera jitter - scene movement

ysraja2415, 2023-11-08:
Hi Team ,
We are using aca1920 155 uc cameras for our machine vision applications. We are recording 12 bit data at high FPS (as our application demands) and converting to BGR format using cv2.COLOR_BayerRGGB2BGR. 

1. There is a small camera jitter introduced which is causing entire scene movement which can be noticed when we magnify the small motions . This happens even after fixing the camera on a tripod and using vibration dampening pads and  some times even when there is no ground vibration at all. Could you please help us in resolving this issue .
2. After starting grabbing we want to drop all the frames that are captured in the first 3 seconds irrespective of resultant fps and we should use one by one grab strategy as per our requirement. Is there any way in pypylon to do it ?

Please let us know on any queries.

Regards,
Raja

TianliangWang, 2023-11-09:
I faced the same issue! and the quality of images is different from the data that acquired by Pylon viewer.

HighImp, 2023-11-13:
Hi Raja,

For the second point, I might have an idea: you can activate the timestamp chunks, and drop all frames until the difference in timestamps between image n and 1 is greater than 3sec. 

I'm not sure what you mean by movement. Do you see a difference before and after the conversion/save or does the image randomly wobble a little?

HighImp, 2023-12-04:
Hi, I have written a small example of discarding images based on the timestamp. 
I think it's best to contact support for your jitter issue as I don't think you can fix it through software

```python
def discard_images(camera: py.InstantCamera, discard_time_ms: int = 3000) -> int:
    """Discard images until the frame start of the upcoming image is <discard_time_ms> later than the first image

    :param camera: Instant Camera Object
    :param discard_time_ms: Minimum time distance between first and last discarded image
    :return: number of images that were discarded
    """

    assert camera.IsGrabbing()  # this is only to discard images, the grabbing have to be started already
    first_timestamp_ms = None
    discarded_images = 0
    while True:
        with camera.RetrieveResult(1000) as grab_res:
            grab_res.BslChunkTimestampSelector.SetValue("FrameStart")
            timestamp_ms = grab_res.BslChunkTimestampValue.Value / 1e6  # timestamps are in ns
        if first_timestamp_ms is None:
            first_timestamp_ms = timestamp_ms
        elif timestamp_ms - first_timestamp_ms > discard_time_ms:
            break
        discarded_images += 1
    return discarded_images

camera = py.InstantCamera(py.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

# activate timestamps
camera.ChunkModeActive.Value = True
camera.ChunkSelector.Value = "Timestamp"
camera.ChunkEnable.Value = True

discard_time_ms = 3000

print(f"Estimated dropped Frames: {round(camera.BslResultingTransferFrameRate.Value*discard_time_ms/1000,1)}")
camera.StartGrabbing(py.GrabStrategy_OneByOne)  # Start grabbing before discarding
print(f"Dropped Frames: {discard_images(camera=camera, discard_time_ms=3000)}")
```