basler/pypylon #680: The issue of image saving

TianliangWang, 2023-11-09:
Hi there,

I am using Python to capture images automatically, but after I compared the data I collected, I found that the data collected by the automatic collection program was different from the data collected using pylon viewer software, and some details seemed to be smoothed out.
My camera model is **Basler acA640-750uc;**
In fact, the function I want is very simple. I just collect data at the highest sampling rate under different exposure times and continue to collect data for two seconds. I originally used the "recording" function in the software to manually sample.
this is the software setting. 
[image]

Below is my code. I am looking forward to receiving any suggestions from you.

```
from pypylon import pylon
import cv2
import os
import threading
import time
import imageio
from PIL import Image
import numpy as np

# set folder to save image
output_folder = os.path.join("Z:/m1a/AG-Fluiddynamic/UserData/T_Wang/Data_Speckle/", "Captured_Images_8")
os.makedirs(output_folder, exist_ok=True)

# build a camera example
tl_factory = pylon.TlFactory.GetInstance()

# obtain all cameras
devices = tl_factory.EnumerateDevices()
if len(devices) == 0:
    print("cannot find the camera")
else:
    for device in devices:
        print(device.GetFriendlyName())

# select the first camera
camera = pylon.InstantCamera(tl_factory.CreateDevice(devices[0]))

# open the camera
camera.Open()

# set the mode to continuous mode
camera.AcquisitionMode.SetValue('Continuous')

# off the auto gain mode
camera.GainAuto = "Off"

# set gain as 0
camera.Gain.SetValue(0)

# set the aquisition speed to max
camera.AcquisitionFrameRateEnable.SetValue(True)
camera.AcquisitionFrameRate.SetValue(camera.AcquisitionFrameRate.GetMax())

camera.Width.SetValue(480)  #
camera.Height.SetValue(480)  #

# set image format
camera.PixelFormat.SetValue('BGR8')

# set bandwidth / camera.BandwidthControlMode = 'Manual'/BayerBG10

# set bandwidth camera.BandwidthControlParameter = 50

# set Buffer Size
camera.MaxNumBuffer = 10000  # set values

# set a task that responsible for image saving
def save_images(exposure_time, images):
    exposure_folder = os.path.join(output_folder, f"Exposure_{exposure_time}")
    os.makedirs(exposure_folder, exist_ok=True)

    for i, cv_image in enumerate(images):
        cv2.imwrite(os.path.join(exposure_folder, f"image_{i}.tiff"), cv_image)

try:
    for exposure_time in range(60, 100, 100):
        # set exposure time
        camera.ExposureTime.SetValue(exposure_time)  # us

        # start to sample data
        camera.StartGrabbing()

        start_time = time.time()
        frame_count = 0
        images = []

        while time.time() - start_time < 2:  # the time for acquisition
            grab_result = camera.RetrieveResult(500, pylon.TimeoutHandling_ThrowException)

            if grab_result.GrabSucceeded():
                # convert image from Pylon to OpenCV (RGB)
                image = pylon.PylonImage()
                image.AttachGrabResultBuffer(grab_result)
                cv_image = image.GetArray()

                cv_image = np.array(cv_image, dtype=np.int8)

                images.append(cv_image)

                frame_count += 1

            grab_result.Release()

        # stop sample
        camera.StopGrabbing()

        # print sample information
        print(f"Exposure Time: {exposure_time} us, Frames Captured: {frame_count}")

        # save images
        save_thread = threading.Thread(target=save_images, args=(exposure_time, images))
        save_thread.start()
        save_thread.join()

except Exception as e:
    print("Error occurred:", e)

# close camera
camera.Close()

# remind finish
print("Acquisition finished")
```

HighImp, 2023-11-13:
Hi, 

i see that you are using the image saving methods from cv2. 
Can you please try the pylon build in image saving?

```python
res = cam.GrabOne(1000)
img = py.PylonImage()
img.AttachGrabResultBuffer(res)
img.Save(py.ImageFileFormat_Tiff, "image.tiff")
```

TianliangWang, 2023-11-13:
> Hi,
> 
> i see that you are using the image saving methods from cv2. Can you please try the pylon build in image saving?
> 
> ```python
> res = cam.GrabOne(1000)
> img = py.PylonImage()
> img.AttachGrabResultBuffer(res)
> img.Save(py.ImageFileFormat_Tiff, "image.tiff")
> ```

OK, Thank you so much! I will try it!

closed 2023-11-13 by TianliangWang