basler/pypylon #684: How resolve this problem?

Wiktor010, 2023-11-14:
[image]

thiesmoeller, 2023-11-14:
Install swig

sudo apt install swig

Wiktor010, 2023-11-15:
I installed swig but the problem is still the same

thiesmoeller, 2023-11-15:
Have you updated pip ?
Is swig really in your path now?

Wiktor010, 2023-11-15:
The pip version is 23.3.1, siwg is in the path /home/pi/.local/bin/swig. Pypylon is in /home/pi/Wiktor/pypylon. When I use pip install . this suggests that swig should be in /tmp/pip-build-env-_21feqdb/overlay/lib/python3.9/site-packages/swig/data/bin/swig and I don't know if this is the exact path where swig should be located

Wiktor010, 2023-11-15:
The pylon is located here
[image]

thiesmoeller, 2023-11-15:
Maybe pip install gets lost somehow on your 32bit environment.

You can also build via 

python setup.py bdist_wheel
Or just. ... build to check that it works

Wiktor010, 2023-11-15:
I still have problems with the swig. Message:RuntimeError: swig executable not found on path! The system installed it in the default location /home/pi/.local/bin/swig. Is this location good? Do I need to change it because there is the same error every time?

thiesmoeller, 2023-11-15:
have you called `swig` from your shell? 
... It might be, that the swig version that gets distributed with pypi is not working on your platform ...

thiesmoeller, 2023-11-15:
As a faster solution to your issue:

Do you want to work on the codebase of pypylon itself? 
In this case you have to compile the swig binary on your own ( swig.org ) , as the pypi version of swig is too new for your system and the version on `apt get swig` is too old.

Or do you want to _use_ pypylon.
In this case you can simple call `pip install pypylon` and get a precompiled version from pypi.

Wiktor010, 2023-11-15:
Will this problem regarding swig be solved on the 64-bit version (aarch64), because I work on Raspberry Pi4 and can reinstall the system? Now I'm working on the 32 bit version and that's why the above problems occur. I'm trying to create a vision system and I also need libraries such as opencv and torch.

thiesmoeller, 2023-11-15:
If you don't need the large address space of a 64bit system for your application I wouldn't make the switch to 64bit system.

It would also be solved, if you run the latest 32bit version of rasbian ( bookworm -> https://www.raspberrypi.com/software/operating-systems/ ) 

The swig package, that you then install via `apt install swig` is matching our requirements.
https://packages.debian.org/bookworm/swig

And also the swig, that you can install via `pip install swig` should work. ( bookworm has newer c-libs)

Wiktor010, 2023-11-15:
Well Install Raspberry Pi OS (32-bit) Debian Bookworm Port with Raspberry Pi Desktop 1.2GB

closed 2023-11-17 by Wiktor010