basler/pypylon #689: connecting the Basler ace acA2440-35um camera with Raspberry Pi4 (bullsey 64-bit)

Wiktor010, 2023-11-19:
I have a problem with connecting the Basler ace acA2440-35um camera with Raspberry Pi4 (bullsey 64-bit). I have a question whether it is necessary to install the SDK from the official Basler website, e.g. version (pylon 7.4.0 Camera Software Suite Linux x86 (64-bit) - Debian installation package) to properly connect to the camera and additionally use the following functionalities:

camera. Preset light sources. SetValue(pylon.LightSourcePreset_Off)
camera.BalanceRatioSelector.SetValue(pylon.BalanceRatioSelector_Red),
camera.BalanceRatio.SetValue(2.0), camera.ReverseX.SetValue(True)
camera.ReverseX.SetValue(True)
camera.ReverseY.SetValue(True)

The next question is what is the easiest way to use the wrapper to connect to the camera and be able to use, among others, of the above functionalities. I used the pip3 install pypylon binary, the library was installed, but I still have no connection to the camera. I tried to clone the repository and used the pip install command. , but unfortunately there is a problem with the pylon paths. Thank you in advance for your reply as I have an important project to do and I'm stuck for a week because of it. If possible, I would like to ask you to include step-by-step instructions in your response, which will allow for the fastest installation and trouble-free connection to the camera.

Generally, what I mean is whether the wrapper itself allows you to use the above functionalities and enables proper connection to the camera without having to install the SDK from the official Basler website.

thiesmoeller, 2023-11-19:
Start systematic:

Install via 

`pip install pip --upgrade`

And 

`pip install pypylon --upgrade`

Is your camera+cable  running with pylon in a PC environment?

How do you connect to the PI4? Direct usb 3 port or via a hub.

What is the output of 
`lsusb`

Disconnect / connect your camera 
`dmesg | tail -30`

Is your script running if you run as root ?

Wiktor010, 2023-11-20:
[image]
the script still does not run the sudo command

thiesmoeller, 2023-11-20:
Please install the full SDK including Basler pylonviewer on your platform.

It seems, that the usb device setup code was not executed properly, and prevents accessing the camera from your user account.

Can you retry to execute

`sudo <pylon_install_location>/share/pylon/setup-usb.sh`

Wiktor010, 2023-11-20:
I have already installed the full SDK installer but the problem is still the same

thiesmoeller, 2023-11-20:
What is the output of the above script invocation ?

thiesmoeller, 2023-11-22:
Is it working now ?

closed 2023-11-22 by Wiktor010