basler/pypylon #691: Basler acA3088-57um has framerate limit on pypylon

parkhy0106, 2023-11-24:
This is my source code

`from pypylon import pylon
import cv2
import numpy as np
import time

camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.Open()

# demonstrate some feature access
new_width = camera.Width.Value - camera.Width.Inc
if new_width >= camera.Width.Min:
    camera.Width.Value = new_width

camera.MaxNumBuffer = 100
camera.StreamGrabber.MaxTransferSize = 4*1024*1024

camera.AcquisitionFrameRateEnable.SetValue(True)  # 프레임 레이트 기능 활성화
camera.AcquisitionFrameRate.SetValue(300.0)  # 예시로 30fps로 설정

current_bandwidth_control_mode = camera.GetNodeMap().GetNode('DeviceLinkThroughputLimitMode')

# Set the bandwidth control mode to 'On'
current_bandwidth_control_mode.SetValue('On')

# Set the desired bandwidth
desired_bandwidth = 419430400
camera.GetNodeMap().GetNode('DeviceLinkThroughputLimit').SetValue(int(desired_bandwidth))

camera.ExposureTime.SetValue(100)
# Set Gain to zero and turn of exposure auto
camera.GainAuto.SetValue('Off')
camera.Gain.SetValue(20)

camera.Width.Value = 376
camera.Height.Value = 376

camera.
camera.MaxNumBuffer = 20000
camera.StartGrabbing()
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

fpss = [0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,
0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,
0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,
0,0,0,0,0]

while camera.IsGrabbing():
    start_time = time.time()
    grabResult = camera.RetrieveResult(1000, pylon.TimeoutHandling_ThrowException)
    if grabResult.GrabSucceeded():
        # Access the image data.
        #print("SizeX: ", grabResult.Width)
        #print("SizeY: ", grabResult.Height)
        image = converter.Convert(grabResult)
        img = image.GetArray()
        #cv2.imshow('test', img)
        #print("Gray value of first pixel: ", img[0, 0])

        k = cv2.waitKey(1)

        elapsed_time = time.time() - start_time
        fps = 1 / elapsed_time
        fpss.append(fps)
        del fpss[0]
        #print(fpss)
        print(f"FPS: {np.average(fpss):.2f}")

        if k == 27:
            break
    grabResult.Release()

camera.Close()
cv2.destroyAllWindows()
`

and the result was like this.
[image]

In pylon-viewer, the same setting can export almost 300fps and it is the exact setting.
I tried every possible way to get higher fps but it doesn't matter what I tried.

Please. I need you guys to help.

HighImp, 2023-11-25:
Hi Parkhy,

can you add

```python
print(camera.BslResultingTranferFrameRate.Value) 
```
before you start grabbing and tell me the value?

If it is higher than your measured loop frequency, 
then may your loop code is delaying the retrieval of the image. 
I suspect the OpenCV WaitKey function. ([see opencv forum](https://answers.opencv.org/question/52774/waitkey1-timing-issues-causing-frame-rate-slow-down-fix/))

SMA2016a, 2024-03-31:
most properly, the delay is caused by doing image conversion in grabloop.  

image = converter.Convert(grabResult)

closed 2024-03-31 by SMA2016a