basler/pypylon #692: How to find the reason why no images are grabbed

maschbauerkl, 2023-11-28:
Hey there,

I have the following problem:
I have a rather complex application inside several docker containers. Inside one container, a celery task is taking pictures with a GigE cam. The pictures are then analyzed and processed further in other containers. At the moment, I am taking pictures with the following line of code/grabstrategy, after initializing the cam object:
img = cam.GrabOne(1000).Array

Everything works fine inside the docker container on my test machine. I now wanted to switch to another, more powerful machine, but can't grab any pictures there. After several attempts, I run into "Runtime Exception: Camera device physically removed". At the moment, I want to understand, why the Array has the length of 0, to get the error. 
When comparing the working docker-compose and the not working docker-compose, everything seems to be the same, network is the same, both pcs are in the same physical network, can access the camera via GigE. Do you have any idea how to troubleshoot this problem?

HighImp, 2023-11-29:
Hello Maschbauerkl, 
can you please execute this sample code and tell the output?

```python
import pypylon.pylon as py
import numpy as np

camera = py.InstantCamera(py.TlFactory.GetInstance().CreateFirstDevice())

try:
    camera.Open()

    print(f"Camera Model Name: {camera.DeviceModelName.Value}")
    print(f"Expected Image Shape: {(camera.Height.Value, camera.Width.Value)}")

    with camera.GrabOne(1000) as grab_result:
        assert grab_result.IsValid(), "Invalid Grab Result!"
        assert grab_result.GrabSucceeded(), f"{grab_result.GetErrorCode()} / {grab_result.GetErrorDescription()}"
        print(f"Resulting Image Shape: {np.shape(grab_result.Array)}")
        print(f"Minimum Pixel Value: {np.min(grab_result.Array)}")
        print(f"Maximum Pixel Value: {np.max(grab_result.Array)}")
finally:
    camera.Close()
```

If your system works correct, the output should look like this:
```
Camera Model Name: a2A2600-64umBAS
Expected Image Shape: (2128, 2600)
Resulting Image Shape: (2128, 2600)
Minimum Pixel Value: 0
Maximum Pixel Value: 5
```
Otherwise, can you tell the error in the assertion?

maschbauerkl, 2023-12-01:
Thank you. This is what I got_
Exception has occurred: AssertionError
3774873620 / The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e. network adapter, switch, or ethernet cable. To fix this, try increasing the camera's Inter-Packet Delay in the Transport Layer category to reduce the required bandwidth, and adjust the camera's Packet Size setting to the highest supported frame size.

The strange thing is, the computer next to it is working tine with the same operationg system. I guess there is something wrong with the computers network settings, I will look into that.

SMA2016a, 2024-04-05:
is the problem solved?

maschbauerkl, 2024-04-08:
Sorry for not closing it. I reinstalled linux and the issue is gone. I think it was some strange behavior due to unknown reasons. :)

closed 2024-04-08 by maschbauerkl