basler/pypylon #696: Problems with packaging basler python development as an exe

ljjcccc, 2023-12-11:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Traceback (most recent call last):
  File "main.py", line 13, in <module>
  File "C:\Users\q1223\AppData\Local\Temp\embedded.hvcu_o0d.zip\shibokensupport\__feature__.py", line 142, in _import
  File "PyInstaller\loader\pyimod02_importers.py", line 419, in exec_module
  File "pypylon\pylon.py", line 37, in <module>
ImportError: cannot import name '_pylon' from 'pypylon' (D:\safe check\demo\output\main\_internal\pypylon\__init__.pyc)

What should I do next?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

64 bit windows 11,basler acA1600-60gm

### Runtime information:

```shell
python: 3.10.13 | packaged by conda-forge | (main, Oct 26 2023, 18:01:37) [MSC v.1935 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 3.0.1rc1 / 7.4.0.38864
```

thiesmoeller, 2023-12-11:
How do you call pyinstaller ?
Have you installed the pyinstaller hooks?

ljjcccc, 2023-12-11:
> How do you call pyinstaller ? Have you installed the pyinstaller hooks?

I I tried the following command to invoke pyinstaller to auto-package the exe:

`pip install pyinstaller auto-py-to-exe `
`auto-py-to-exe`

but it failed

ljjcccc, 2023-12-11:
> It seems to be an issue with changes in pyinstaller >= 6.0.0 ( [pyinstaller/pyinstaller#7619](https://github.com/pyinstaller/pyinstaller/pull/7619) )
> 
> You might test using a 5.x version of pyinstaller.

Successfully resolved the problem, thank you very much for your help！ @thiesmoeller

thiesmoeller, 2023-12-11:
you can use latest pyinstaller by using the updated pyinstaller-hooks-contrib, 
we'll check to fix this upstream. 

`pip install git+https://github.com/thiesmoeller/pyinstaller-hooks-contrib.git@feature/fix_hook_pypylon_pyi_6.x`