basler/pypylon #699: Low processing speed of images in Ubuntu than Mac

RamyaNarayanan-Git, 2023-12-15:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,
I'm using pypylon to grab images from basler camera and storing the images in a file. But this is storing broken / half-loaded images in ubuntu. The time interval for grabbing each image is set 200 ms in the python program. But when I use macbook the images are fine and speed is fast. Since, we are planning for ubuntu setup can you suggest a way to resolve or troubleshoot this issue in ubuntu ?

Regards,
Ramya

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

camera used: Basler USB 3.0 acA2440-35uc
Ubuntu version:  20.04 
Architecture type: x86_64
python 3.8
pypylon: 2.2.0-cp3.8 whl

### Runtime information:

```shell
from pypylon import pylon
import time
import cv2
import requests
import base64
img = pylon.PylonImage()

# conecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly) 

print("Using device ", camera.GetDeviceInfo().GetModelName())
converter = pylon.ImageFormatConverter()

# converting to opencv bgr format
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    
    if grabResult.GrabSucceeded():
        # Access the image data
        image = converter.Convert(grabResult)
        img = image.GetArray()
        # image_base64 = base64.b64encode(grabResult.tobytes()).decode('utf-8')
        # print(image_base64)
        time.sleep(0.5)
        cv2.imwrite("image", img, [cv2.IMWRITE_JPEG_QUALITY, 80])
        files = {'image': open("image", 'rb')} 
        # requests.post(node_url, files=files)
        # time.sleep(0.5)

        with open("image", "rb") as image_file:
            encoded_string = base64.b64encode(image_file.read())
        # encoded_string = base64.b64encode(str(img))
        # encoded_string = str(img).encode('base64')
        # print(encoded_string)

        
        payload ={"filename": "image.png", "filedata": encoded_string}
        resp = requests.post("my_url", data=payload)
        # time.sleep(0.5)

    grabResult.Release()
    
# Releasing the resource    
camera.StopGrabbing()
```

thiesmoeller, 2023-12-15:
some comments:
you don't have to write the file to disk, which is expensive.
* cv2 has the imencode function to do jpg encoding from array to array

you don't have to mime encode your data. 
* HTTP is capable of transferring binary data. Request library supports this, see: https://requests.readthedocs.io/en/latest/user/advanced/#post-multiple-multipart-encoded-files

```python
import requests

# Specify the URL for the POST request
url = 'https://example.com/upload'

...
_, jpg_data = cv2.imencode(".jpg", img_array)

# Additional fields
additional_data = {
    'filename': 'image.jpg',
}

# Send the file and additional fields in the POST request as part of a form
response = requests.post(url, files={'file_data': jpg_data}, data=additional_data)

# Handle the response as needed
print(response.text)
``` 

check if this already helps to speed up your code.

RamyaNarayanan-Git, 2023-12-26:
Thank you so much for your response! I'll try them out

closed 2024-05-31 by SMA2016a