basler/pypylon #700: Read operation on device '2676:ba02:2:4:44' failed: 'The requested operation timed out.' : TimeoutException thrown (file 'PylonUsbDevice.cpp', line 778)

DuyLocHoang, 2023-12-22:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am using two basler acA1920-25uc camera and run it on Jetson Orin NX 16Gb and .
In two days time, Consistently I am getting the following error.
Read operation on device '2676:ba02:2:4:44' failed: 'The requested operation timed out.' : TimeoutException thrown (file 'PylonUsbDevice.cpp', line 778) in one of the two basler cameras
The frequency of errors is not frequent.
Do anyone know what is root cause to this error and how to fix it?
'''

class Camera :
    # Config Camera
    def __init__(self,serial_number,nodeFile = None):
        self.nodeFile = nodeFile
        self.camera = self.connect_camera(serial_number)
        self.camera.Open()
        
        print("Using device ", self.camera.GetDeviceInfo().GetModelName())
        print("Reading file back to camera's node map...")
        # Load settings from node file
        self.loadSettings(self.nodeFile)
        # Start Grab Image
        self.camera.TriggerMode.SetValue('On')
        # self.camera.TriggerDelay.SetValue(0)
        # self.camera.TriggerSelector.SetValue('FrameStart')
        self.camera.TriggerSource.SetValue('Software')        # 
        self.startGrab()
        
    def loadSettings(self,nodeFile):
        # load settings from Node file
        if nodeFile :
            pylon.FeaturePersistence.Load(nodeFile, self.camera.GetNodeMap(), True)
    
    def startGrab(self):
        # Start Grab Image
        self.camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

    def stopGrab(self):
        # Stop Grab Image
        self.camera.StopGrabbing()

    def connect_camera(self,serial_number):
        # Connect and check Camera
        devices = pylon.TlFactory.GetInstance().EnumerateDevices()
        for device in devices:
            try :
                camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateDevice(devices[1]))
                # CHECK IF CAMERA 1 IS CORRECT SERIAL NUMBER
                

                if camera.GetDeviceInfo().GetSerialNumber() == serial_number :
                    print("CAM1-info: ",camera.GetDeviceInfo().GetSerialNumber())
                    return camera
            except Exception as error :
                print("CAMERA",error)
    
    def triggerCamera(self):
        # Capture camera
        # if self.camera.WaitForFrameTriggerReady(10000):
        time.sleep(0.005)
        try :
            if self.camera.WaitForFrameTriggerReady(5000, pylon.TimeoutHandling_ThrowException) :
                self.camera.ExecuteSoftwareTrigger()
                grabResult = self.camera.RetrieveResult(1000, pylon.TimeoutHandling_ThrowException) 
                if grabResult.GrabSucceeded():
                    img = grabResult.Array
                    return img
        except Exception as error:
            print("Trigger Error: ",error)
        return None

    def saveImage(self,grabResult,typeImage,quality = 100 ):
        # Save image into folder
        now = datetime.now()
        dateTime = now.strftime("%d-%m-%Y_%H-%M-%S")
        img = pylon.PylonImage()
        img.AttachGrabResultBuffer(grabResult)
        if typeImage == "jpg" :
            ipo = pylon.ImagePersistenceOptions()
            ipo.SetQuality(quality)
            filename = f"{dateTime}.jpg"
            img.Save(pylon.ImageFileFormat_Jpeg, filename, ipo)
        else :
            filename = f"{dateTime}.png"
            img.Save(pylon.ImageFileFormat_Png, filename)
'''

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

PC : Jetson Orin NX 16GB (Seeed Studio)
Connect Camera through USB3.0 
Cables : 2200000621 (basler cables)

### Runtime information:

```shell
pypylon: 2.3.0 / 7.2.1.19268
```

SMA2016a, 2024-01-06:
What action is needed to get the camera in good shape back?

-just restarting your application?
-system reboot?
-just removing/plug in  the usb cable?

DuyLocHoang, 2024-03-27:
I just restarting my application

DuyLocHoang, 2024-03-27:
Then I switched to using hardware triggers via I/O. I encountered a new problem. Normally, the time to trigger an image takes about 50ms.
However, after operating for a while, there will be some trigger images about 3-4ms (Time measurement via PLC). And these photos are just black images.
Can you help me evaluate the above case.
Equipment : 
Camera : aca1920-25uc
PLC : Keyence KV-7500
I/O Cable : 2000034087