basler/pypylon #701: Different behaviour between Windows and Linux computer

guanggevwu, 2023-12-22:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I have two computers: one Ubuntu and one windows. I have four sets of code to connect to a camera. They all work in Windows but some don't work on Ubuntu.
1.  This one works on Linux computer. However, normally I want to connect to a camera with a certain IP address. 
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

2. This one doesn't work on Linux.   
factory = pylon.TlFactory.GetInstance()  
ptl = factory.CreateTl('BaslerGigE')  
empty_camera_info = ptl.CreateDeviceInfo()  
empty_camera_info.SetPropertyValue('IpAddress', '141.213.13.66')  
camera_device = factory.CreateFirstDevice(empty_camera_info)  
# or CreateDevice doesn't work either.
camera = pylon.InstantCamera(camera_device)  

3. This one doesn't work on Linux.  
factory = pylon.TlFactory.GetInstance()  
for dev_info in factory.EnumerateDevices():  
        if dev_info.GetDeviceClass() == 'BaslerGigE' and dev_info.GetIpAddress() == "141.213.13.66":  
            camera = pylon.InstantCamera(factory.CreateDevice(dev_info))  

4. This one works on Linux by using "CreateFirstDevice".  
factory = pylon.TlFactory.GetInstance()  
for dev_info in factory.EnumerateDevices():  
        if dev_info.GetDeviceClass() == 'BaslerGigE' and dev_info.GetIpAddress() == "141.213.13.66":  
            camera = pylon.InstantCamera(factory.CreateFirstDevice(dev_info))  

So basicallt it seems 4 is the only way to go. I believe this is not intended.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Both Linux and Windows. Camera Basler acA640-121gm

### Runtime information:

```shell
I have pylon viewer installed on Windows and I have pylon_7.4.0.14900-deb0_amd64.deb installed on Linux.
```

SMA2016a, 2023-12-25:
use universal device object instead of using GigE one.

Try test this code:

hostname = "141.213.13.66"
tlf = pylon.TlFactory.GetInstance()
ptl = pylon.TlFactory.GetInstance()
empty_camera_info = pylon.DeviceInfo()
empty_camera_info.SetPropertyValue('IpAddress', hostname)
camera_device = tlf.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)

guanggevwu, 2024-01-03:
Thanks. This does work. Is there a reason to write " ptl =
pylon.TlFactory.GetInstance()" since ptl is never used after it is defined?

SMA2016a, 2024-01-04:
@guanggevwu 

just a mistake form my side. Simply remove this line

closed 2024-02-01 by SMA2016a