basler/pypylon #702: how select camera by number or name?

sagregrevc, 2024-01-02:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I saw some starter examples, but even for multiple cameras they never use number or name

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

-

### Runtime information:

```shell
-
```

sagregrevc, 2024-01-02:
probably finded answer 
https://github.com/basler/pypylon/issues/63

thiesmoeller, 2024-01-02:
Hi @sagregrevc, 

please take a look into the documentation and sample project repository:  https://github.com/basler/pypylon-samples

To your specific question:
Which "name" do you mean?

The process to filter for a device is by setting up empty `DeviceInfo` objects. On these objects you can set e.g. a DeviceSerialNumber.

On the subsequent call to EnumerateDevices you'll only get your device matching the serial number.
( https://docs.baslerweb.com/pylonapi/cpp/struct_pylon_1_1_i_device_factory#function-enumeratedevices_1 and https://docs.baslerweb.com/pylonapi/cpp/class_pylon_1_1_c_device_info)

Quick sample:
```python
import pypylon.pylon as py

tlf = py.TlFactory.GetInstance()
```

```python
# list first five available cameras
for d in tlf.EnumerateDevices()[:5]:
    print("----")
    # typical filter rules:
    print(d.GetDeviceClass())
    print(d.GetModelName())
    print(d.GetSerialNumber())
    print(d.GetUserDefinedName())
    # other filter rules
    print(d.GetFullName())
    print(d.GetFriendlyName())

```

    ----
    BaslerGigE
    acA5000-25gc
    202350

    Basler acA5000-25gc#00305332470D#192.168.0.50:3956
    Basler acA5000-25gc (202350)
    ----
    BaslerGigE
    acA5000-25gc
    202351

    Basler acA5000-25gc#003053335683#192.168.0.51:3956
    Basler acA5000-25gc (202351)
    ----
    BaslerGigE
    acA5000-25gc
    202352

    Basler acA5000-25gc#0030533445BE#192.168.0.52:3956
    Basler acA5000-25gc (202352)
    ----
    BaslerGigE
    acA5000-25gc
    202353

    Basler acA5000-25gc#003053354453#192.168.0.53:3956
    Basler acA5000-25gc (202353)
    ----
    BaslerGigE
    acA5000-25gc
    202354

    Basler acA5000-25gc#003053367347#192.168.0.54:3956
    Basler acA5000-25gc (202354)

```python
# select specific camera by serial number
di = py.DeviceInfo()
# all options set here are ANDed
di.SetSerialNumber("202353")
di.SetDeviceClass("BaslerGigE")

# enumerate using a list of deviceinfo entries
# in case of multiple entries the entries are ORed

for d in tlf.EnumerateDevices([ di , ]):
    print("----")
    # typical filter rules:
    print(d.GetDeviceClass())
    print(d.GetModelName())
    print(d.GetSerialNumber())
    print(d.GetUserDefinedName())
    # other filter rules
    print(d.GetFullName())
    print(d.GetFriendlyName())

```

    ----
    BaslerGigE
    acA5000-25gc
    202353

    Basler acA5000-25gc#003053354453#192.168.0.53:3956
    Basler acA5000-25gc (202353)

closed 2024-02-17 by SMA2016a