basler/pypylon #703: How to query Stream Grabber Statistics Parameters from pypylon?

cmoranda, 2024-01-04:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am attempting to get the Statistics_Resend_Request_Count parameter defined in the Stream Grabber Parameters defined [here](https://docs.baslerweb.com/stream-grabber-parameters). I cannot seem to find any sample documentation that interacts with these statistics parameters, but I assume there must be a why to get them in Python? Thanks!

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

pypylon running on PC:

-  X86_64
- Windows 10

-Camera connected to ethernet hub

### Runtime information:

```shell
pypylon 2.3.0
pylon 7.3
```

SMA2016a, 2024-01-04:
camera.Open() 
camera.StartGrabbingMax(5) 

# Display the Total Buffer count value
camera.StreamGrabber.Statistic_Total_Buffer_Count.GetValue() 
print("total buffer count " + str(value)) 

# Display the Buffer underrun count value
camera.StreamGrabber.Statistic_Buffer_Underrun_Count.GetValue() 
print("total bufferrun count " + str(value)) 

# Display the Resend Request count value
camera.StreamGrabber.Statistic_Resend_Request_Count.GetValue() 
print("total resend count" + str(value)) 

# Display the GetQueuedBuffer count value = camera.GetQueuedBufferCount() 
print("total GetQueuedBufferCount " + str(value)) 

# Stop the camera acquisition and close 
camera.StopGrabbing() 
camera.Close()

cmoranda, 2024-01-04:
This is just what I was looking for, thank you!

closed 2024-01-08 by SMA2016a