basler/pypylon #704: BUG: Sample grabusinggrabloopthread.py rarely works

atomicmac, 2024-01-09:
### Describe the issue:

I am attempting to run the code sample grabusinggrabloopthread.py. However, it only works about 1/7 times.
There are no error messages displayed when it does not work, it just seems like the image event callbacks are never getting called. The config events are handled fine.
It also seems necessary to power down the camera before it will work again.
I've tried running it with and without debugger, results are the same.
Output from working / non-working is provided below.
I have also attempted wrapping the event handlers in try / except blocks to get more info out of exceptions that could be getting thrown in the sample handlers, but I have never seen a successful run with the try / except wrappers. I have included the version of the sample code with try/ except blocks here.

### Reproduce the code example:

```python
# Grab_UsingGrabLoopThread.cpp
# This sample illustrates how to grab and process images using the grab loop thread
# provided by the Instant Camera class.
import time
import traceback

from pypylon import genicam
from pypylon import pylon

# Contains a Configuration Event Handler that prints a message for each event method call.

class ConfigurationEventPrinter(pylon.ConfigurationEventHandler):
    def OnAttach(self, camera):
        print("OnAttach event")

    def OnAttached(self, camera):
        print("OnAttached event for device ", camera.GetDeviceInfo().GetModelName())

    def OnOpen(self, camera):
        print("OnOpen event for device ", camera.GetDeviceInfo().GetModelName())

    def OnOpened(self, camera):
        print("OnOpened event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabStart(self, camera):
        print("OnGrabStart event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabStarted(self, camera):
        print("OnGrabStarted event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabStop(self, camera):
        print("OnGrabStop event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabStopped(self, camera):
        print("OnGrabStopped event for device ", camera.GetDeviceInfo().GetModelName())

    def OnClose(self, camera):
        print("OnClose event for device ", camera.GetDeviceInfo().GetModelName())

    def OnClosed(self, camera):
        print("OnClosed event for device ", camera.GetDeviceInfo().GetModelName())

    def OnDestroy(self, camera):
        print("OnDestroy event for device ", camera.GetDeviceInfo().GetModelName())

    def OnDestroyed(self, camera):
        print("OnDestroyed event")

    def OnDetach(self, camera):
        print("OnDetach event for device ", camera.GetDeviceInfo().GetModelName())

    def OnDetached(self, camera):
        print("OnDetached event for device ", camera.GetDeviceInfo().GetModelName())

    def OnGrabError(self, camera, errorMessage):
        print("OnGrabError event for device ", camera.GetDeviceInfo().GetModelName())
        print("Error Message: ", errorMessage)

[... 125 lines cut ...]
CSampleImageEventHandler::OnImageGrabbed called.

Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) e
e

Process finished with exit code 139 (interrupted by signal 11:SIGSEGV)
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

acA4024-8gm connected to X86/X86_64 Ubuntu computer over ethernet
Netgear switch
Cat-6 ethernet cable, about 8' between PoE injector and camera, and 6' between PoE injector and switch

### Runtime information:

```shell
python: 3.11.4 (main, Jul  8 2023, 14:19:04) [GCC 11.3.0]
platform: linux/x86_64/6.1.0-1027-oem
pypylon: 3.0.1 / 7.4.0.38864
```

thiesmoeller, 2024-01-10:
please call your sample using debugger to see from where the segmentation violation is triggered:

`gdb python sample.py`

in gdb call `run` to start the program.

when you are stopped in the segmentation violation.

call `bt` to output the backtrace of the error and copy this output here.

atomicmac, 2024-01-10:
There does not appear to be any segfault? Here is the output of the gdb session:

`(gdb) run
Starting program: /home/mike/.pyenv/versions/3.11.4/envs/ampcv-3.11.4/bin/python3.11 sample.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffefbff640 (LWP 114339)]
[New Thread 0x7fffed3fe640 (LWP 114340)]
[New Thread 0x7fffecbfd640 (LWP 114341)]
[New Thread 0x7fffea3fc640 (LWP 114342)]
[New Thread 0x7fffe7bfb640 (LWP 114343)]
[New Thread 0x7fffe53fa640 (LWP 114344)]
[New Thread 0x7fffe2bf9640 (LWP 114345)]
[New Thread 0x7fffe03f8640 (LWP 114346)]
[New Thread 0x7fffdbbf7640 (LWP 114347)]
[New Thread 0x7fffd93f6640 (LWP 114348)]
[New Thread 0x7fffd6bf5640 (LWP 114349)]
[New Thread 0x7fffd43f4640 (LWP 114350)]
[New Thread 0x7fffd1bf3640 (LWP 114351)]
[New Thread 0x7fffd13f2640 (LWP 114352)]
[New Thread 0x7fffcebf1640 (LWP 114353)]
[New Thread 0x7fffca3f0640 (LWP 114354)]
[New Thread 0x7fffc7bef640 (LWP 114355)]
[New Thread 0x7fffc53ee640 (LWP 114356)]
[New Thread 0x7fffc2bed640 (LWP 114357)]
[New Thread 0x7fffbd7ff640 (LWP 114358)]
[New Thread 0x7fffbcffe640 (LWP 114359)]
[New Thread 0x7fffb7fff640 (LWP 114361)]
[Thread 0x7fffb7fff640 (LWP 114361) exited]
OnOpen event for device  acA4024-8gm
[New Thread 0x7fffb7fff640 (LWP 114363)]
[Thread 0x7fffcebf1640 (LWP 114353) exited]
[Thread 0x7fffc2bed640 (LWP 114357) exited]
[Thread 0x7fffc53ee640 (LWP 114356) exited]
[Thread 0x7fffc7bef640 (LWP 114355) exited]
[Thread 0x7fffca3f0640 (LWP 114354) exited]
[Thread 0x7fffd13f2640 (LWP 114352) exited]
[Thread 0x7fffd1bf3640 (LWP 114351) exited]
[Thread 0x7fffd43f4640 (LWP 114350) exited]
[Thread 0x7fffd93f6640 (LWP 114348) exited]
[Thread 0x7fffe53fa640 (LWP 114344) exited]
[Thread 0x7fffe7bfb640 (LWP 114343) exited]
[Thread 0x7fffea3fc640 (LWP 114342) exited]
[Thread 0x7fffecbfd640 (LWP 114341) exited]
[Thread 0x7fffed3fe640 (LWP 114340) exited]
[Thread 0x7fffefbff640 (LWP 114339) exited]
[Thread 0x7fffd6bf5640 (LWP 114349) exited]
[Thread 0x7fffdbbf7640 (LWP 114347) exited]
[Thread 0x7fffe03f8640 (LWP 114346) exited]
[Thread 0x7fffe2bf9640 (LWP 114345) exited]
[Detaching after fork from child process 114364]
[New Thread 0x7fffc2bed640 (LWP 114365)]
OnOpened event for device  acA4024-8gm
OnGrabStart event for device  acA4024-8gm
[New Thread 0x7fffc53ee640 (LWP 114366)]
[New Thread 0x7fffc7bef640 (LWP 114367)]
[New Thread 0x7fffca3f0640 (LWP 114368)]
[New Thread 0x7fffefbff640 (LWP 114369)]
[New Thread 0x7fffea3fc640 (LWP 114370)]
OnGrabStarted event for device  acA4024-8gm
Thread Name: MainThread
Thread ID: 140737352756096
Thread Alive: True
Thread Daemon: False
Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) t
t
Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) t
t
Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) t
t
Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) e
e
[Thread 0x7fffea3fc640 (LWP 114370) exited]
[Thread 0x7fffefbff640 (LWP 114369) exited]
[Thread 0x7fffca3f0640 (LWP 114368) exited]
[Thread 0x7fffc53ee640 (LWP 114366) exited]
[Thread 0x7fffc2bed640 (LWP 114365) exited]
[Thread 0x7fffb7fff640 (LWP 114363) exited]
[Thread 0x7fffbcffe640 (LWP 114359) exited]
[Thread 0x7fffbd7ff640 (LWP 114358) exited]
[Thread 0x7ffff7eaeb80 (LWP 114334) exited]
[Thread 0x7fffc7bef640 (LWP 114367) exited]
[New process 114334]
[Inferior 1 (process 114334) exited normally]`

atomicmac, 2024-01-10:
Also, here is the output from a gdb session which successfully called the image event callbacks:

Starting program: /home/user/.pyenv/versions/3.11.4/envs/cv-3.11.4/bin/python3.11 sample.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffefbff640 (LWP 116081)]
[New Thread 0x7fffed3fe640 (LWP 116082)]
[New Thread 0x7fffecbfd640 (LWP 116083)]
[New Thread 0x7fffe83fc640 (LWP 116084)]
[New Thread 0x7fffe5bfb640 (LWP 116085)]
[New Thread 0x7fffe33fa640 (LWP 116086)]
[New Thread 0x7fffe2bf9640 (LWP 116087)]
[New Thread 0x7fffe03f8640 (LWP 116088)]
[New Thread 0x7fffddbf7640 (LWP 116089)]
[New Thread 0x7fffdb3f6640 (LWP 116090)]
[New Thread 0x7fffd8bf5640 (LWP 116091)]
[New Thread 0x7fffd63f4640 (LWP 116092)]
[New Thread 0x7fffd3bf3640 (LWP 116093)]
[New Thread 0x7fffcf3f2640 (LWP 116094)]
[New Thread 0x7fffccbf1640 (LWP 116095)]
[New Thread 0x7fffcc3f0640 (LWP 116096)]
[New Thread 0x7fffc9bef640 (LWP 116097)]
[New Thread 0x7fffc73ee640 (LWP 116098)]
[New Thread 0x7fffc2bed640 (LWP 116099)]
[New Thread 0x7fffbd7ff640 (LWP 116100)]
[New Thread 0x7fffbcffe640 (LWP 116101)]
[New Thread 0x7fffb7fff640 (LWP 116107)]
[Thread 0x7fffb7fff640 (LWP 116107) exited]
OnOpen event for device  acA4024-8gm
[New Thread 0x7fffb7fff640 (LWP 116108)]
[Thread 0x7fffc2bed640 (LWP 116099) exited]
[Thread 0x7fffc9bef640 (LWP 116097) exited]
[Thread 0x7fffccbf1640 (LWP 116095) exited]
[Thread 0x7fffcf3f2640 (LWP 116094) exited]
[Thread 0x7fffd3bf3640 (LWP 116093) exited]
[Thread 0x7fffc73ee640 (LWP 116098) exited]
[Thread 0x7fffd63f4640 (LWP 116092) exited]
[Thread 0x7fffcc3f0640 (LWP 116096) exited]
[Thread 0x7fffd8bf5640 (LWP 116091) exited]
[Thread 0x7fffddbf7640 (LWP 116089) exited]
[Thread 0x7fffdb3f6640 (LWP 116090) exited]
[Thread 0x7fffe03f8640 (LWP 116088) exited]
[Thread 0x7fffe2bf9640 (LWP 116087) exited]
[Thread 0x7fffe33fa640 (LWP 116086) exited]
[Thread 0x7fffe5bfb640 (LWP 116085) exited]
[Thread 0x7fffe83fc640 (LWP 116084) exited]
[Thread 0x7fffecbfd640 (LWP 116083) exited]
[Thread 0x7fffed3fe640 (LWP 116082) exited]
[Thread 0x7fffefbff640 (LWP 116081) exited]
[Detaching after fork from child process 116109]
[New Thread 0x7fffc2bed640 (LWP 116110)]
OnOpened event for device  acA4024-8gm
OnGrabStart event for device  acA4024-8gm
[New Thread 0x7fffc73ee640 (LWP 116111)]
[New Thread 0x7fffc9bef640 (LWP 116112)]
[New Thread 0x7fffcc3f0640 (LWP 116113)]
[New Thread 0x7fffefbff640 (LWP 116114)]
[New Thread 0x7fffe83fc640 (LWP 116115)]
OnGrabStarted event for device  acA4024-8gm
Thread Name: MainThread
Thread ID: 0x7ffff7eaeb80
Thread Alive: True
Thread Daemon: False
Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) t
t
OnImageGrabbed event for device  acA4024-8gm
Thread Name: Dummy-1
Thread ID: 0x7fffe83fc640
Thread Alive: True
Thread Daemon: True
SizeX:  4024
SizeY:  3036
Gray values of first row:  [0 0 0 ... 0 0 0]

CSampleImageEventHandler::OnImageGrabbed called.

Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) t
t
OnImageGrabbed event for device  acA4024-8gm
Thread Name: Dummy-1
Thread ID: 0x7fffe83fc640
Thread Alive: True
Thread Daemon: True
SizeX:  4024
SizeY:  3036
Gray values of first row:  [0 0 0 ... 0 0 0]

CSampleImageEventHandler::OnImageGrabbed called.

Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) t
t
OnImageGrabbed event for device  acA4024-8gm
Thread Name: Dummy-1
Thread ID: 0x7fffe83fc640
Thread Alive: True
Thread Daemon: True
SizeX:  4024
SizeY:  3036
Gray values of first row:  [0 0 0 ... 0 0 0]

CSampleImageEventHandler::OnImageGrabbed called.

Enter "t" to trigger the camera or "e" to exit and press enter? (t/e) e
e
[Thread 0x7fffe83fc640 (LWP 116115) exited]
[Thread 0x7fffefbff640 (LWP 116114) exited]
[Thread 0x7fffcc3f0640 (LWP 116113) exited]
[Thread 0x7fffc73ee640 (LWP 116111) exited]
[Thread 0x7fffc2bed640 (LWP 116110) exited]
[Thread 0x7fffb7fff640 (LWP 116108) exited]
[Thread 0x7fffbcffe640 (LWP 116101) exited]
[Thread 0x7fffbd7ff640 (LWP 116100) exited]
[Thread 0x7ffff7eaeb80 (LWP 116080) exited]
[Thread 0x7fffc9bef640 (LWP 116112) exited]
[New process 116080]
[Inferior 1 (process 116080) exited normally]

atomicmac, 2024-01-11:
I swapped the camera to an acA4024-8gc and it works every time.
Any idea what the difference is that would make the acA4024-8gm only get the image callbacks working occasionally and require a camera power cycle to get a successful run again?

BTW, I tried the 3.0.1, 2.3.0 and 2.2.0 versions of the pypylon library with the same results on the acA4024-8gm.

thiesmoeller, 2024-01-11:
Is the the acA4024-8gm working in pylon viewer when enable trigger set the source to software and push the trigger software button ?

atomicmac, 2024-01-11:
Interesting, so while the "single shot" GUI button works in pylon viewer, clicking the "Generate Software Trigger" Execute button does not seem to do anything.

[image]

thiesmoeller, 2024-01-11:
To test keep your trigger configuration and start continuous streaming ( equivalent to start streaming in pypylon)

atomicmac, 2024-01-11:
I hit the "Continuous Shot" button in the GUI with the same configuration as above and nothing happens.

thiesmoeller, 2024-01-12:
To recap:
Configure  in pylon viewer:

* to be sure the camera is in factory settings:
   * `UserSetSelector` = "Default"
   * push `UserSetLoad` button
* your trigger config:
  * `TriggerSelector` = FrameStart"
  * `TriggerMode` = "On"
  * `TriggerSource` = "Software"

In the viewer push the "Continuous Shot" button 

And now push the `TriggerSoftware` feature to trigger one frame each time you push

atomicmac, 2024-01-13:
Thank you, ok following those above instructions I get a frame when I push the Generate Software Trigger button.

atomicmac, 2024-01-13:
Adding

`    cam.Open()
    cam.UserSetSelector.Value = "Default"
    cam.UserSetLoad.Execute()`

before the `camera.RegisterConfiguration(pylon.SoftwareTriggerConfiguration(), pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_Delete)` command makes the sample code work.

I was thinking that registering the SoftwareTriggerConfiguration with RegistrationMode_ReplaceAll would replace all of the previous configuration, but some configuration persists through that call?

thiesmoeller, 2024-01-13:
Great!

The Configurations that you can load will just add up upon what is already configured on the camera. 

Good hint from your issue to better document how prior configuration (dynamic/userset) can alter the behaviour of a pypylon script.

atomicmac, 2024-01-16:
Thanks for your help Thies! I was looking for what SoftwareTriggerConfiguration does change, but can't find it documented anywhere - could you point me to the proper documentation?
Thanks again!