basler/pypylon #705: Poor acA2440-20gc perforance on Jetson Orin

tc360950, 2024-01-10:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,
I’m trying to use Basler acA2440-20gc GIGE camera on Jetson Orin with PoE.
With frame size set to ~ 600x600 I can achieve at least 60FPS, but whenever I increase frame size (e.q. to 2000x2000) even 1FPS is too much and the camera disconnects (error: Connection lost "Basler acA2440-20gc, there are no other errors in the logs). The issue is present both on Pylon Viewer and when using pypylon.

The camera works properly on another Jetson device (Xavier). It also has no problems working on my vanilla laptop. MTU on the interface is set to 9000.
Driver for the Xavier is:

driver: igb
version: 5.10.120-tegra

whilst Orin has:
driver: lan743x
version: 5.10.104-tegra

I' using pylon Viewer 7.2.1.

Do you have any suggestions how can I diagnose this issue?

Thank you for all suggestions.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

NVIDIA Jetson Orin NX Jetpack 5.1.1
CPU 8-core Arm® Cortex®-A78AE v8.2 64- bit CPU; 2MB L2 + 4MB L3 
Ethernet 4* LAN RJ45 GbE PoE(PSE 802.3 af 15 W, 10/100/1000Mbps)

### Runtime information:

```shell
pyton Viewer 64-Bit

pyton Release 7.2.1.25747
pylon C++ SDK 7.2.1.19268
pylon Application 1.4.0.11689
pylon Viewer 7.2.0.11689

python: 3.10.13 (main, Dec 19 2023, 10:51:32) [GCC 12.2.0]
platform: linux/aarch64/5.10.104-tegra
pypylon: 3.0.1 / 7.4.0.38864
```

tc360950, 2024-01-11:
Camera works properly when powered externally (i.e. not through PoE)

closed 2024-01-11 by tc360950

thiesmoeller, 2024-01-11:
To learn from your issue, what was the brand and modelof POE injector that you have been using?