basler/pypylon #706: BUG: The size of saved images unmatched

SnowCharmQ, 2024-01-16:
### Describe the issue:

Hi, 
I am trying to record images when using the Basler camera. However, I noticed some of these images have the size 1920\*1200, and some have the size 1831\*1138 when the image format is **tiff** or **bmp**. When the image format is **jpeg**, the problem can be solved but the resolution of the image is not good enough.

### Reproduce the code example:

```python
from pypylon import pylon
from datetime import datetime

tl_factory = pylon.TlFactory.GetInstance()
devices = tl_factory.EnumerateDevices()
camera = pylon.InstantCamera(tl_factory.CreateDevice(devices[0]))
camera.Open()
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)

converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_RGB8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned
while camera.IsGrabbing():
    grab_result = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    img_time = datetime.now().strftime('%Y%m%d%H%M%S%f')
    img = converter.Convert(grab_result)
    img.Save(pylon.ImageFileFormat_Tiff, str(img_time) + '.tiff')
    grab_result.Release()
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

X86
Windows 10
USB3.0 Connection

### Runtime information:

```shell
python: 3.10.13 | packaged by Anaconda, Inc. | (main, Sep 11 2023, 13:24:38) [MSC v.1916 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 3.0.1 / 7.4.0.38864
```

SMA2016a, 2024-01-17:
Could you please print the width and height from the grab result and verify whether these values are also being reported incorrectly?

SnowCharmQ, 2024-01-20:
> Could you please print the width and height from the grab result and verify whether these values are also being reported incorrectly?

Sorry for the late reply. 
I have already tried to print the width and height of the grab result, they are all the same size 1920\*1200, but their corresponding saved images are 1831\*1138 sometimes.

SMA2016a, 2024-01-21:
would you tell me the camera model that you are using,please? 

In adaption will you check the grab result if it is grabbed successful before save, 

 if grabResult.GrabSucceeded():

thiesmoeller, 2024-01-21:
... And what tool do you use to determine the size of the image?

SnowCharmQ, 2024-01-22:
> would you tell me the camera model that you are using,please?
> 
> In addtion will you check the grab result if it is grabbed successfully before save,
> 
> if grabResult.GrabSucceeded():
> 
> @SnowCharmQ will you also upload an image with size (1831*1138)

> ... And what tool do you use to determine the size of the image?

[image]
The camera I used can be seen from the image. I have tried to add `if grabResult.GrabSucceeded():`, the result is still the same.
In the program, I just use the print function `print(img.GetWidth(), img.GetHeight())` to determine the size. After saving, I use WPS Image to view images.

SnowCharmQ, 2024-01-22:
I think I have already discovered the problem. It is the bug of WPS Image. When I continuously select the next image, the software may compress the image size during loading due to the large size of the pictures. However, when I directly select a specific image, its size is correct. I think maybe I need to open an issue to them :(

Thank you for your kind help!

closed 2024-01-25 by SnowCharmQ