basler/pypylon #709: Release schedule for Python 3.12

BenediktBurger, 2024-01-23:
### Describe the issue:

No wheel present for Python 3.12.

When do you plan to release a wheel for Python 3.12?

### Reproduce the code example:

```python
-
```

### Error message:

```shell
-
```

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup & camera model(s) used

Python 3.12

### Runtime information:

```shell
-
```

m-binev, 2024-03-05:
@BenediktBurger Hi, we will do that togehter with the new pylon 7.5 release, e.g., in May, 24. 
Regards

closed 2024-03-05 by m-binev

BenediktBurger, 2024-03-05:
Thanks for letting me know.

Eivinddo, 2024-04-09:
@m-binev , our team is working on upgrading our systems to python3.12 and the only missing dependency is pypylon.
Do you have any unstable/unfinished release or branch we can checkout and test before the pylon 7.5 release in May?

m-binev, 2024-04-09:
Hi @Eivinddo  , thank you for working on that - much appreciated indeed. We are still in the process of preparing/releasing pylon 7.5 and cannot offer you any beta version yet. Thus, I'd ask you for some more patience, please.

Eivinddo, 2024-04-09:
Thank you for the quick reply and information!
We look forward to the next release:)

Jedw11, 2024-04-29:
@m-binev , can you give an idea of when in May pylon 7.5 will be released?

BenediktBurger, 2024-04-29:
Hey @Jedw11 , he stated previously that it will be released on May, 24th.

Eivinddo, 2024-05-30:
Hi again! Sorry to bother you @m-binev, but I see that Pylon 7.5 has been released last week. Do you have an estimate for when Pypylon will release a new version that supports 7.5 and Python 3.12?

m-binev, 2024-05-30:
@Eivinddo Hi, you don't bother at all)) 
You're right, we've just released pylon 7.5 and we are now on the way to start working on that task either. Because our backend (framework, API) has changed a little bit, we might need a little more time to get it working fine in pypylon. In a nutshell, we are working on that & I'd like to ask you for a little more patience. Regards

Eivinddo, 2024-05-30:
Okay, thanks!