basler/pypylon #710: Problem with the sequencer to grab images for HDR

larssommerlade, 2024-01-23:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi there,

I have a problem with the Code mentioned in #67. I'm using a a2A1920-160ucPRO. I always gets the Error:

Exception has occurred: LogicalErrorException
Node not existing (file 'genicam_wrap.cpp', line 16504)
  File "C:\Users\sommerlade\Documents\Basler_HDR\main.py", line 33, in <module>
    cam.SequencerMode.Value = "Off"
_genicam.LogicalErrorException: Node not existing (file 'genicam_wrap.cpp', line 16504)

Code:
```python
from pypylon import pylon as py
import numpy as np

# open the first USB device
info = py.DeviceInfo()
info.SetDeviceClass("BaslerUsb")
cam = py.InstantCamera(py.TlFactory.GetInstance().CreateFirstDevice(info))

# open device
cam.Open()

cam.UserSetSelector = "Default"
cam.UserSetLoad.Execute()

# mean exposure time 20ms
exp_0 = 2000
exp_1 = 20000
exp_2 = 200000

# enable camera chunk mode
cam.ChunkModeActive = True
# enable exposuretime chunk
cam.ChunkSelector = "ExposureTime"
cam.ChunkEnable = True

cam.SequencerMode.Value = "Off"
cam.SequencerConfigurationMode.Value = "On"

cam.SequencerSetSelector = 0

cam.ExposureTime = exp_0

cam.SequencerSetSave.Execute();

cam.SequencerSetSelector = 1

cam.ExposureTime = exp_1

cam.SequencerSetSave.Execute();

cam.SequencerSetSelector.Value = 2

cam.ExposureTime = exp_2

# select that we jump to set 0 after this set
# path 1 is the _next_ path / path 0 would be _reset_ path
cam.SequencerPathSelector = 1
cam.SequencerSetNext = 0

cam.SequencerSetSave.Execute();

cam.SequencerConfigurationMode.Value = "Off"

# this will set the first sequencer set to set _0_ as side effect
cam.SequencerMode.Value = "On"

# grab 4 sets of 3 images
cam.StartGrabbingMax( 4 * 3)

while cam.IsGrabbing():
    res = cam.RetrieveResult(1000)
    exp_time_chunk = res.ChunkDataNodeMap.ChunkExposureTime.Value
    print("%d\t%6.0f\t%6.2f" %(res.BlockID, exp_time_chunk, np.mean(res.Array)))
    # ... do something with the images
    
    # return to buffer queue
    res.Release()
    
cam.StopGrabbing()

cam.SequencerMode = "Off"

cam.Close()

```

Have someone an idea where the problem is?

Many Thanks

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Camera model: a2A1920-160ucPRO

PC:
Microsoft Windows 10
RAM: 64GB
CPU: Intel(R) Xeon(R) CPU E3-1270 v5

### Runtime information:

```shell
python: 3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 3.0.1 / 7.4.0.38864
```

HighImp, 2024-01-24:
Hi, 

Please change your write access calls from:
cam.<NodeName> = <new_value>
 
to:
cam.<NodenName>.Value = <new_value>

Best regards

SMA2016a, 2024-01-24:
@larssommerlade  did you check with pylonviewer if your camera has got the sequencer parameter? if not, please post the firmware version of the camera.

larssommerlade, 2024-01-25:
@SMA2016a in the pylonviewer I didnt't find a sequencer parameter. 
My Firmware Version is:  p=vu3p_imx392c_pro/s=r/v=2.1.0/i=6518.29/h=e035835

SMA2016a, 2024-01-26:
would you contact your local support team to get a firmware update for your camera?

larssommerlade, 2024-01-30:
@SMA2016a many thanks for your help, I have updated it and it worked

closed 2024-01-30 by larssommerlade