basler/pypylon #711: The relationship bewteen the frame rate and the exposure time

RealBrandonChen, 2024-01-26:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I set the frame rate of my camera with 

```
cam.AcquisitionFrameRateEnable.Value = True
cam.AcquisitionFrameRate.Value = 125
d = cam.ResultingFrameRate.Value
```

and the printed frame rate is 125 exactly, but when I use code to set the exposure time:
```
cam.ExposureTime.Value = 8000 # 1/8000 * 1e6 for frame rate
d = cam.ResultingFrameRate.Value
```
the resulting frame rate is 124.2, lower than 125.
I tried other frame rates and exposure time, it turns out that the resulting frame rate would always be lower than ideal rate when I set it using cam.ExposureTime.Value = pictureTime.
I was wondering what's the relationship between the frame rate and the exposure time for the Basler camera.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Camera type: aca640-750um

### Runtime information:

```shell
python: 3.11.7 | packaged by Anaconda, Inc. | (main, Dec 15 2023, 18:05:47) [MSC v.1916 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 3.0.0 / 7.4.0.38864
```

SMA2016a, 2024-01-30:
Your question is not related to pypylon. The behavior you're experiencing is due to the camera hardware itself. Kindly reach out to your local support team or distributor to discuss this issue in more detail

closed 2024-04-13 by SMA2016a

thiesmoeller, 2024-04-13:
What you experience is called Frame Overhead Time.
The time the image sensor is prepared for the next frame.