basler/pypylon #712: pylon-libusb-1.0.so causing segfault in our application

mahiuchun, 2024-01-30:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I noticed that in both pylon and pypylon for Linux, the package contains `pylon-libusb-1.0.so` which is a build of [libusb](https://github.com/libusb/libusb) 1.0.

It's unclear to me which minor version of libusb 1.0 is used or if there is any Basler specific modification 

FWIW, libusb 1.0.0 was released in 2008, it's still being updated as 1.0.x, currently at 1.0.26

Currently the presence of `pylon-libusb-1.0.so` causes segfault in our applications when there are other users of libusb 1.0.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

x86-64 PC Ubuntu 20.04/22.04
 
acA2440-35uc USB3

### Runtime information:

```shell
pypylon: 1.8.0 / 6.3.0.18933
```

SMA2016a, 2024-01-31:
the minor version is  19

cseurope, 2024-02-11:
@mahiuchun did you solve the issue?

mahiuchun, 2024-02-12:
No I did not. I am trying to create a small program to reproduce; it's going to take a while.

cseurope, 2024-02-13:
OK. Thanks for the update