basler/pypylon #713: pypylon causing application segfault when a different version of pylon viewer is installed

mahiuchun, 2024-01-30:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

pypylon (we use 1.8.0 from pip) does not need a pylon viewer installation to function.

However, we noticed that when pylon viewer (we use DEB package) is installed. The python program would end up loading shared libraries from `/opt/pylon` (we looked at process memory maps).

When the pypylon version and pylon viewer version are too far apart, our Python application would end up segfault or exhibit other instability.

Is there a way to achieve isolation between pypylon and system pylon viewer?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

x86-64 PC Ubuntu 20.04/22.04

acA2440-35uc USB3

### Runtime information:

```shell
pypylon: 1.8.0 / 6.3.0.18933
```

SMA2016a, 2024-02-01:
try to unset the environment variable PYLON_ROOT

mahiuchun, 2024-02-02:
> try to unset the environment variable PYLON_ROOT

Thanks for the response. Now I noticed that pylon DEB package contains `/etc/profile.d/basler-gentl-path.sh` and `/etc/profile.d/basler-pylon-root.sh`.

Un-setting `GENICAM_GENTL64_PATH` made the difference for me. Un-setting `PYLON_ROOT` had no effect