basler/pypylon #715: BUG: Segmentation Fault

omerts, 2024-01-31:
### Describe the issue:

When running tests, or trying to run on a camera, I am getting a segmentation fault:
```
Running suite(s): states_good
0%: Checks: 3, Failures: 0, Errors: 3
../tests/check/generic/states.c:114:E:general:test_state_changes_up_and_down_seq:0: (after this point) Received signal 11 (Segmentation fault)
../tests/check/generic/states.c:146:E:general:test_state_changes_up_seq:0: (after this point) Received signal 11 (Segmentation fault)
../tests/check/generic/states.c:178:E:general:test_state_changes_down_seq:0: (after this point) Received signal 11 (Segmentation fault)
Check suite states ran in 16.217s (tests failed: 3)
```

### Reproduce the code example:

```python
Just run: `ninja -C builddir test`
```

### Error message:

```shell
ninja: Entering directory `builddir'
[1/2] Running all tests.
1/1 generic_states        FAIL            16.23s   exit status 3
>>> UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 MALLOC_PERTURB_=156 CK_DEFAULT_TIMEOUT=20 GST_PLUGIN_SYSTEM_PATH_1_0='' GST_PLUGIN_LOADING_WHITELIST=gstreamer:gst-plugins-base:gst-plugin-pylon@/home/amram/development/gst-plugin-pylon/builddir GSETTINGS_BACKEND=memory GST_STATE_IGNORE_ELEMENTS='
' GST_PLUGIN_PATH_1_0=/home/amram/development/gst-plugin-pylon/builddir:/usr/lib/x86_64-linux-gnu/gstreamer-1.0:/usr/lib/x86_64-linux-gnu/gstreamer-1.0 GST_REGISTRY=/home/amram/development/gst-plugin-pylon/builddir/tests/check/generic_states.registry GST_PLUGIN_SCANNER_1_0=/usr/lib/x86_64-linux-gnu/gstreamer1.0/gstreamer-1.0/gst-plugin-scanner ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 /home/amram/development/gst-plugin-pylon/builddir/tests/check/generic_states
――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――
Running suite(s): states_good
0%: Checks: 3, Failures: 0, Errors: 3
../tests/check/generic/states.c:114:E:general:test_state_changes_up_and_down_seq:0: (after this point) Received signal 11 (Segmentation fault)
../tests/check/generic/states.c:146:E:general:test_state_changes_up_seq:0: (after this point) Received signal 11 (Segmentation fault)
../tests/check/generic/states.c:178:E:general:test_state_changes_down_seq:0: (after this point) Received signal 11 (Segmentation fault)
Check suite states ran in 16.217s (tests failed: 3)
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――

Summary of Failures:

1/1 generic_states FAIL            16.23s   exit status 3

Ok:                 0   
Expected Fail:      0   
Fail:               1   
Unexpected Pass:    0   
Skipped:            0   
Timeout:            0   

Full log written to /home/amram/development/gst-plugin-pylon/builddir/meson-logs/testlog.txt
FAILED: meson-internal__test 
/usr/local/bin/meson test --no-rebuild --print-errorlogs
ninja: build stopped: subcommand failed.
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

This is happening both on my personal Ubuntu 22.04 computer, and on our Jetson Xavier NX Jetpack 5.1.2.

CPU: X86_64 & ARM64
Operating system: Ubuntu 20.04 & 22.04
RAM: 32Gb & 8Gb
Cable length: 25cm
Deepstream: 6.4
Cuda: 12.3 & 11.4

### Runtime information:

```shell
python: 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
platform: linux/x86_64/5.17.15-051715-generic
pypylon: 3.0.1 / 7.4.0.38864
```

thiesmoeller, 2024-01-31:
Can you post in gst-plugin-pylon project and delete here?

omerts, 2024-02-01:
@thiesmoeller sorry had both tabs open, closing...

closed 2024-02-01 by omerts