basler/pypylon #716: _genicam.AccessException: Node is not writable.

kesaroid, 2024-01-31:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am trying to reduce the resolution of images grabbed on the camera end using either [Scaling](https://docs.baslerweb.com/scaling), [Pixel beyond](https://docs.baslerweb.com/pixel-beyond#pixel-beyond), or [Binning](https://docs.baslerweb.com/binning).

I have seen most of the previous issues regarding this matter was related to camera not being open. However, I have the cameras open and am able to set exposure time and framerate settings. When I try to use any one of the following 
```
camera.ScalingHorizontal.Value = 3 # scaling
camera.BinningHorizontal.Value = 2 # Binning
```
It gives me Node not existing error, which matches with basler documentation mentioning my camera version does not have these features.
However, Pixel beyond feature exists with the camera, and when I try the following `camera.BslScalingFactor.Value = 3 # Pixel beyond` It gives me a Node is not writable error.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Camera: a2A1920-51gcPRO

### Runtime information:

```shell
File "camera_bandwidth_test.py", line 43, in init_camera                                                                  camera = _set_camera_attributes(camera)                                                                               File "camera_bandwidth_test.py", line 13, in _set_camera_attributes                                                       camera.BslScalingFactor.Value = 3 # Pixel beyond                                                                      File "D:\ML\env\airsv2\lib\site-packages\pypylon\genicam.py", line 2086, in SetValue                                      return _genicam.IFloat_SetValue(self, Value, Verify)                                                                _genicam.AccessException: Node is not writable. : AccessException thrown in node 'BslScalingFactor' while calling 'BslScalingFactor.SetValue()' (file 'floatt.h', line 85)
```

SMA2016a, 2024-01-31:
did you open the camera before accessing the camera parameter?

most color cameras do not have the binning :  BinningHorizontal  please check with pylon viewer if both parameters are available.

if not, turn to your distributor to get the firmware update.

HighImp, 2024-01-31:
If your camera is open and the error persists: 
Pixel Beyond features require a debayered image, but a Bayer8 image is active by default. 
Please change your pixel format to RGB/BGR or YCbCr and try again.

kesaroid, 2024-02-01:
> did you open the camera before accessing the camera parameter?
> most color cameras do not have the binning : BinningHorizontal please check with pylon viewer if both parameters are available.

As mentioned above, the cameras are opened before setting the parameters such as exposure time and framerate. 
ScalingHorizontal and BinningHorizontal settings don't seem to exist on pylon viewer, which makes sense for pypylon to throw me a `_genicam.LogicalErrorException: Node not existing` error. 
But BslScalingFactor on pylon shows up as `<not available>` and pypylon throws a `Node not writable.` error.

kesaroid, 2024-02-01:
> If your camera is open and the error persists: Pixel Beyond features require a debayered image, but a Bayer8 image is active by default. Please change your pixel format to RGB/BGR or YCbCr and try again.

Upon using `camera.PixelFormat.SetValue('RGB8')` I am able to use the Pixel Beyond feature. 
I am able to successfully reduce the camera resolution on the camera end.
Thank you.

closed 2024-02-01 by kesaroid