basler/pypylon #718: 3791651346 Payload data has been discarded, (but only infrequently)

matkir, 2024-02-06:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

```
def get_frame(self, idx: int, image_path: str) -> None:
            self.cameras[idx].ExecuteSoftwareTrigger()
            with self.cameras[idx].RetrieveResult(500) as grab_result:
                if grab_result.GrabSucceeded():
                    frame = self.converter.Convert(grab_result)
                    cv2.imwrite(image_path, frame, [int(cv2.IMWRITE_JPEG_QUALITY), 95])  # type: ignore
                else:
                    error_code = str(grab_result.ErrorCode)
                    print(error_code, grab_result.ErrorDescription)  <--- print from this further down
                    #we retry this code after 0.01 seconds
```
Using Basler dart daa3840.
This code is run multiple times every minute, at random intervals, only capturing ONE image at a time.
When an image is not GrabSucceeded(), we will retry the image capture up to 5 times. 
If it first fails, it will seldom capture it on the next 5 tries either.
Running usb3.1, with 10gig on the hub.
Also worth noting that we are running 3 GiGE cameras at full speed at the same time. but we are not CPU limited.
This is a problem on multiple machines, so it should not me hardware failure. 

What we have tried
* lowering framerate
* lowering max transfer size

```
3791651346 Payload data has been discarded. Payload data can be discarded by the camera device if the available bandwidth is insufficient.
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Nvidia Jestson Xavier, Ubuntu 18, 32G ram. 

### Runtime information:

```shell
python: 3.6.9 (default, Mar 10 2023, 16:46:00) 
[GCC 8.4.0]
platform: linux/aarch64/4.9.140-l4t-r32.4
pypylon: 2.3.0 / 7.2.1.19268
```

SMA2016a, 2024-02-07:
try to set following values

camera.DeviceLinkThroughputLimit.setvalue = 160000000

camera.StreamGrabber.MaxTransferSize.setvalue = 4*1024*1024

matkir, 2024-02-09:
Our first initial tests show that the problem went from >0.8% of the captures to <0.05%.
Tank you so much for the help.

closed 2024-02-09 by matkir