basler/pypylon #719: GigE PTP synchronization problem using a2A2448-105g5mBAS cameras

DavidSayada, 2024-02-07:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I have two a2A2448-105g5mBAS cameras that are connected on the same Neousys 5GigE (4 Port) Card connected to an Ubuntu 20.04.6 LTS workstation via PCIe. I am trying to perform synchronized image acquisition from the two cameras using PTP.
On the Ubuntu workstation, I am using ptp4l in order to create a grand-master clock, as described below. `enp9s0` is my port connected to the internet, `enp6s0` and `enp8s0` are the Neousys ports where the cameras are connected

`sudo ptp4l -i enp9s0 -i enp8s0 -i enp6s0 -m --priority1 1 -l 7 --logSyncInterval -1`

This command fails because the requested timestamping mode is not supported, `ethtool -T enp8s0` results below.

> Time stamping parameters for enp8s0:
> Capabilities:
> 	software-transmit     (SOF_TIMESTAMPING_TX_SOFTWARE)
> 	software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
> 	software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
> PTP Hardware Clock: none
> Hardware Transmit Timestamp Modes: none
> Hardware Receive Filter Modes: none

Below is what I am getting when running the ptp4l command.

> ptp4l[11341.111]: config item (null).assume_two_step is 0
> ptp4l[11341.111]: config item (null).check_fup_sync is 0
> ptp4l[11341.111]: config item (null).tx_timestamp_timeout is 1
> ptp4l[11341.111]: config item (null).clock_servo is 0
> ptp4l[11341.111]: config item (null).clock_servo is 0
> ptp4l[11341.111]: config item (null).clockClass is 248
> ptp4l[11341.111]: config item (null).clockAccuracy is 254
> ptp4l[11341.111]: config item (null).offsetScaledLogVariance is 65535
> ptp4l[11341.111]: config item (null).productDescription is ';;'
> ptp4l[11341.111]: config item (null).revisionData is ';;'
> ptp4l[11341.111]: config item (null).userDescription is ''
> ptp4l[11341.111]: config item (null).manufacturerIdentity is '00:00:00'
> ptp4l[11341.111]: config item (null).domainNumber is 0
> ptp4l[11341.111]: config item (null).slaveOnly is 0
> ptp4l[11341.111]: config item (null).gmCapable is 1
> ptp4l[11341.111]: config item (null).gmCapable is 1
> ptp4l[11341.111]: config item (null).time_stamping is 1
> ptp4l[11341.111]: config item (null).twoStepFlag is 1
> ptp4l[11341.111]: config item (null).twoStepFlag is 1
> ptp4l[11341.111]: config item (null).time_stamping is 1
> ptp4l[11341.111]: config item (null).priority1 is 1
> ptp4l[11341.111]: config item (null).priority2 is 128
> ptp4l[11341.111]: interface index 2 is up
> ptp4l[11341.111]: interface index 6 is up
> ptp4l[11341.111]: interface 'enp8s0' does not support requested timestamping mode
> failed to create a clock

I am proceeding the correct way? Is there a possibility to change the time stamping mode on the Neousys ports?

Thank you in advance for your help/

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

PC running Ubuntu 20.04.6 LTS, X86_64
a2A2448-105g5mBAS cameras connected on Neousys 5GigE (4 Port) PCIe Card

### Runtime information:

```shell
pylon Viewer 64-Bit
pylon Release 7.2.1.25747
pylon C++ SDK 7.2.1.19268
pylon Application 1.4.0.11689
pylon Viewer 7.2.0.11689
```

thiesmoeller, 2024-02-07:
What is the kernel version on your system?

qeroll, 2024-02-07:
Working with @DavidSayada on this issue. Here's the uname -a output:

Linux Stereo 5.15.0-91-generic #101~20.04.1-Ubuntu SMP Thu Nov 16 14:22:28 UTC 2023 x86_64 x86_64 x86_64

DavidSayada, 2024-02-08:
I tried to enable HW timestamping in PTP using the `hwstamp_ctl` Linux command the following way, unfortunately that doesn't work neither.

**Command**
`sudo hwstamp_ctl -i enp8s0 -r 13 -t 1`

**Results**

> Device driver does not have support for non-destructive SIOCGHWTSTAMP.
> SIOCSHWTSTAMP failed: Operation not supported

Should we enable a specific option at the Kernel level such as indicated below?

```
CONFIG_PTP_1588_CLOCK=y
CONFIG_PTP_1588_CLOCK_DPAA=y
```

DavidSayada, 2024-02-08:
I succeeded in having my lab work using the `ptpd` utility instead the following way, now my two cameras are PTP slaves.

**First camera**

`sudo ptpd -i enp8s0 -EMV --debug`

**Second camera**

`sudo ptpd -i enp6s0 -EMLV --debug`

Does that sound good for you as a possible solution?

thiesmoeller, 2024-02-08:
starting multiple ptp daemons on one system is not recommended.

As a quick fix to get it working with ptp4l is to use the `-S` switch to allow software timestamping.

`ptp4l -i enp9s0 -i enp8s0 -i enp6s0 -m --priority1 1 -l 7 --logSyncInterval -1 -S`

DavidSayada, 2024-02-08:
Working. Thank you so much! I am closing the ticket.

closed 2024-02-08 by DavidSayada