basler/pypylon #721: Support for color calibration

goncalog, 2024-02-15:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Is it possible to use `pypylon` to do color calibration programmatically (instead of using `pylon Viewer`)?

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup & camera model(s) used

Basler daA1920-160uc

### Runtime information:

```shell
python: 3.8.10 (default, Nov 22 2023, 10:22:35)
[GCC 9.4.0]
platform: linux/aarch64/5.10.104-tegra
pypylon: 3.0.1
```

SMA2016a, 2024-02-15:
no, sorry.

HighImp, 2024-02-15:
Hi,

no, the color calibrator is the only way to do the full calibration. 
But with the dart camera, the feature to calibrate, the color transformation matrix and color adjustment, are not available to customize at all. But off  course, with a white patch, you can use auto white balance via software to get the correct white point.

thiesmoeller, 2024-02-16:
Hi @goncalog , what would have been you expected integration in you application?
I'm asking because color calibration the way we provide it, is a GUI assisted automatic process that guides the user to reach the optimum measurement setup.
Or would your application take care to provide the required operating point of the camera and you were just looking for the calculation part of it?

goncalog, 2024-02-16:
thank you both! 

hi @thiesmoeller, we're still in testing phase for a new camera so mostly trying to understand what functionality is available. we currently use a Jetson Orin which we deploy to production mostly without a monitor, so our preferred solution is to run web apps which we access via the browser (after ssh'ing into the device). we use this method to e.g. configure the camera, do colour calibration, and record videos. so with Basler, and for colour calibration, our only option would be a monitor to use your GUI? or could we easily interact with your GUI via the browser?

thiesmoeller, 2024-02-16:
if you have a ssh channel to your camera you can easily use remote X11 to start the pylonviewer.

start the ssh connection with `ssh -X <jetson_ip>` from a X11 machine.
If you now start pylon, the GUI will be rendered on your local machine.
or you can even call pylonviewer with a VNC frontend using `/opt/pylon/bin/pylonviewer -platform vnc`

m-binev, 2024-03-05:
Hi @goncalog, I think all has been said here (i.e., the pylon Color Calibrator is only available within the plyon Viewer for cameras supporting the given color feature set) and we could close that case. 
Regards

closed 2024-03-05 by m-binev