basler/pypylon #723: BUG: <Please write a comprehensive title after the 'BUG: ' prefix>

Mamooshe, 2024-02-21:
### Describe the issue:

by using pip install pypylon, the root doesn't recognize pypylon library in nvidia jetson board. I need camea grabbing to be executred in autorun and for autorun access to root is required. so all libraries should be defind in root. How can I install basler library in root?

### Reproduce the code example:

```python
pip install pypylon
```

### Error message:

```shell
pypylon is not known in root after sudo su
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

c

### Runtime information:

```shell
c
```

Mamooshe, 2024-02-21:
> sudo pip install pypylon

I tested it but still doesn't know pypylon library in root. If I use :
sudo su
python3
import pypylon

gives me error of not recognizes pypylon

but without sudo su everything is ok and because my program is crontab, it must be executed in root

thiesmoeller, 2024-02-21:
Calling a user process from a `crontab` in Unix-like operating systems involves scheduling a command or script to run at specified times or intervals. Here's a general guide on how to do this:

### 1. Open the Crontab File

First, you need to open the crontab file for editing. You can do this for the current user by running the following command in the terminal:

```sh
crontab -e
```

If you need to edit the crontab for another user (assuming you have the necessary permissions), you can use:

```sh
sudo crontab -u username -e
```

### 2. Add a New Cron Job

In the crontab file, you can add a new line that specifies when and what command you want to execute. The syntax for a cron job is:

```plaintext
* * * * * command-to-execute
```

The asterisks represent different units of time:

- **Minute (0 - 59)**
- **Hour (0 - 23)**
- **Day of the month (1 - 31)**
- **Month (1 - 12)**
- **Day of the week (0 - 6)** (Sunday=0)

For example, to run a script every day at 5:00 PM, you would use:

```plaintext
0 17 * * * /path/to/your/script.sh
```

### 3. Environment and User

When you schedule tasks with crontab, they are run with the user's environment who owns the crontab. It means that environment variables available in your terminal session might not be available for scripts run by cron. Therefore, it's a good practice to define needed environment variables within your script or explicitly in the crontab.

If your script needs to run as a specific user, make sure that user's crontab is the one being edited, or ensure the script switches to the correct user if necessary and possible.

### 4. Troubleshooting

- **Permissions:** Ensure the script or command has executable permissions.
- **Environment Variables:** Scripts run from cron may not have access to the same environment variables as when run manually. It's often helpful to explicitly set or source environment variables in your script.
- **Output and Errors:** To help with debugging, you can redirect the output and errors to a log file:

```plaintext
0 17 * * * /path/to/your/script.sh > /path/to/logfile 2>&1
```

### 5. Validate Changes

After adding your cron job, save and exit the editor. The crontab service will automatically pick up the changes. You can list the current user's cron jobs with:

```sh
crontab -l
```

This setup should be sufficient for most uses of cron to call user processes. Adjustments may be needed based on specific requirements, such as handling more complex scheduling or environment setups.

Mamooshe, 2024-02-22:
> Therefore, it's a good practice to define needed environment variables within your script or explicitly in the crontab.

thank you.
as ou mentioned  
"it's a good practice to define needed environment variables within your script or explicitly in the crontab"
so my environment is root. How can I install pypylon python API in root?
by using pip install pypylon it's not installed in root.

thiesmoeller, 2024-02-22:
The hint with crontab above was, that you can enable running crontab as your user account.

As I don't know what your OS version, python version and username is, here some examples

If you install pypylon as a user jetson. On e.g. Ubuntu 18.04 the install will be in 

`/home/jetson/.local/lib/python3.6/site-packages`

the python interpreter will find this, as this is the std location for users.

you could reference this also from your root crontab by setting

`PYTHONPATH=/home/jetson/.local/lib/python3.6/site-packages/`

in the root environment.

Alternative would be a system install using

```bash
sudo su
```

```bash
pip3 install pip --upgrade
pip3 install pypylon
```

then on ubuntu 18.04 the install will end up in /usr/local/lib/python3.6/dist-packages/

thiesmoeller, 2024-03-08:
Is your issue solved?