basler/pypylon #727: Does StartGrabbing put the camera into free-run mode? Should StartGrabbing not be used with hardware triggers?

atomicmac, 2024-03-01:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

When using "triggered acquisition using the background loop" in this sample [iPython notebook](https://github.com/basler/pypylon-samples/blob/main/notebooks/USB_hardware_trigger_and_chunks.ipynb) as an example for doing hardware-triggered imaged capture using the background grab loop, the call to StartGrabbing appears to put the camera into free-run mode and begins capturing immediately, even though the camera has been configured for hardware capture.
It would seem from the previous example in that file titled "Grabbing with Hardware Trigger on line4" that StartGrabbing should not be used in conjunction with hardware-triggered image capture using a background grab loop - is that correct?
I ask because we're not seeing any captures taking place once configured for hardware trigger and without StartGrabbing.

First reset the camera to Default settings:

```
camera.UserSetSelector.Value = "Default"
camera.UserSetLoad.Execute()
```

Then the camera is configured with the following settings:
```
{
        "ExposureAuto": "Off",
        "ExposureMode": "Timed",
        "ExposureTimeAbs": 29995.0,
        "GainAuto": "Off",
        # Setup the io section
        "LineSelector": "Line1",
        "LineMode": "Input",
        # Setup the trigger / acquisition controls
        "TriggerActivation": "RisingEdge",
        "TriggerMode": "On",
        "TriggerSelector": "FrameStart",
        "TriggerSource": "Line1",
}
```

Then the image event handler is registered:
`camera.RegisterImageEventHandler(pusher_image_event_handler, pylon.RegistrationMode_Append, pylon.Cleanup_Delete)`

When using StartGrabbing, it is invoked like:
`camera.StartGrabbing(pylon.GrabStrategy_OneByOne, pylon.GrabLoop_ProvidedByInstantCamera)`

The image event handler successfully prints out messages and handles the images when it receives them, but that is only happening with StartGrabbing invoked.

We are also currently working on validating that the correct pin is activated over our wiring harness via oscilloscope.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

acA4024-8gc
X86_64 running Ubuntu 22.04

### Runtime information:

```shell
python: 3.11.4 (main, Jul  8 2023, 14:19:04) [GCC 11.3.0]
platform: linux/x86_64/6.1.0-1034-oem
pypylon: 3.0.1 / 7.4.0.38864
```

HighImp, 2024-03-01:
Hi,

Please check if the trigger mode is "On" after StartGrabbing.
If not:
The default ConfigurationEventHandler disable Extertal trigger on OpenCamera. To remove the default handler, enter the following line before configuring the trigger Mode.
`python camera.RegisterConfiguration(py.ConfigurationEventHandler(), py.RegistrationMode_ReplaceAll, py.Cleanup_Delete)
'

Best regards

atomicmac, 2024-03-01:
Wow, thank you @HighImp that worked!
Can't believe this isn't mentioned in the examples or the hardware capture documentation.

closed 2024-03-01 by atomicmac