basler/pypylon #730: RuntimeException thrown (file 'tlfactory.cpp', line 694)

sulaimansiddiqi233, 2024-03-06:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

we just want to run the basler camera via python code but it gives the below error. the camera works in pylon viewer program but it not works in python. 

the code: 
from pypylon import pylon
import cv2
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    # if grabResult.GrabSucceded():
    image = converter.Convert(grabResult)
    img = image.GetArray()

    cv2.imshow("video", img)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
    cv2.destroyAllWindows()
    cv2.waitKey()
  the erorr:
Traceback (most recent call last):
  File "C:\Users\MTS\Desktop\python\test_kamera.py", line 4, in <module>
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
  File "C:\Users\MTS\AppData\Local\Programs\Python\Python39\lib\site-packages\pypylon\pylon.py", line 1545, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'tlfactory.cpp', line 694)

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

from pypylon import pylon
import cv2
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    # if grabResult.GrabSucceded():
    image = converter.Convert(grabResult)
    img = image.GetArray()

    cv2.imshow("video", img)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
    cv2.destroyAllWindows()
    cv2.waitKey()

### Runtime information:

```shell
from pypylon import pylon
import cv2
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.StartGrabbing(pylon.GrabStrategy_LatestImageOnly)
converter = pylon.ImageFormatConverter()
converter.OutputPixelFormat = pylon.PixelType_BGR8packed
converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

while camera.IsGrabbing():
    grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    # if grabResult.GrabSucceded():
    image = converter.Convert(grabResult)
    img = image.GetArray()

    cv2.imshow("video", img)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
    cv2.destroyAllWindows()
    cv2.waitKey()
```

sulaimansiddiqi233, 2024-03-06:
@thiesmoeller @SMA2016a

thiesmoeller, 2024-03-06:
Please edit the sections of issue report
Especially what is your camera and Pylon version and Pypylon version and platform

sulaimansiddiqi233, 2024-03-07:
we are using . pylon viewer 64 bit . pylon release: 7.4.0.14900
camera: acA1300-60gm_23246104
Python : 3.9 64 bit

SMA2016a, 2024-03-11:
would you post the IP and Subnet mask of the camera and NIC,please?

sulaimansiddiqi233, 2024-03-11:
@SMA2016a @thiesmoeller 
it is localhsot ip : 192.168.1.148 and the mask is : 255.255.255.255

SMA2016a, 2024-03-11:
the mask is : 255.255.255.255   ->   use the mask is : 255.255.255.0

sulaimansiddiqi233, 2024-03-11:
@SMA2016a 
we used that mask also. it works in pylon viewer app but not works in pyton code.
 camera: acA1300-60gm_23246104
Python : 3.9 64 bit.
What is NIC?

SMA2016a, 2024-03-11:
IP address and Subnetmack  of camera and network

sulaimansiddiqi233, 2024-03-11:
@SMA2016a 
our camera works in loca. the ip is 192.168.1.84 
mask is : 255.255.255.0
camera: acA1300-60gm_23246104 : Ethernet

EdithJK, 2024-03-13:
Just to get it right - IP address and mask from camera and Network interface card (NIC) adapter

NIC   192.168.1.148 and the mask is : 255.255.255.**255**
CAM 192.168.1.84  and the mask is : 255.255.255.0

Did you change the mask of the **network adapter** from 255.255.255.**255** -> 255.255.255.**0** ?
The mask for the network adapter also has to be 255.255.255.**0**

sulaimansiddiqi233, 2024-03-13:
@EdithJK 
i changed the IPs it works in pylon viewers application but it not works in python run program.

SMA2016a, 2024-03-14:
please create a tcpdmp file.  start tcpdump capture in one console and then start pylon IPconfiguration tool and press 2 times the refresh button. Then close tcpdump and upload the resulting log

sulaimansiddiqi233, 2024-03-14:
finally the camera works with pyhton code. i do not know why it gives errors. so now we can work with it. thank you. 
@SMA2016a @EdithJK @thiesmoeller

closed 2024-03-14 by sulaimansiddiqi233