basler/pypylon #731: Synchronously recording from multiple cameras using PTP

canliu0414, 2024-03-08:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I am trying to set up a system to synchronously record videos/frames from multiple cameras (4 to potentially 12). I have previously attempted to achieve this with USB cameras that we already had, using hardware triggering from arduino. That progress was in another issue: https://github.com/basler/pypylon/issues/671. I think I wasn't able to reach synchronization because the way I saved output in the code was not ideal, maybe a background loop would have solved my problem.

To make this process easier, and also reduce the need of GPIO cables used for synchronization, we decided to switch to ethernet cameras which can be synced via PTP. I am currently testing on several ace2 gigE cameras (like a2A1920-51gmPRO) and using the periodic signal feature for the synchronization.

Here is the piece of code I am working with now, gathered from all over pylon github & issues, to set up PTP between cameras, and acquire frames/videos in a background loop.

```
from pypylon import pylon
import numpy as np
import cv2
import time
import warnings
import threading

framerate = 30.
exposure = 5.
gain = 1.
compression_ratio = 70.
pixel_scaling = 2.79
num_frames1 = 0;
num_frames2 = 0;

# Create VideoWriters for both cameras
fourcc = cv2.VideoWriter_fourcc(*'XVID')
out1 = cv2.VideoWriter('D:\\videoCaptures\\output1.avi', fourcc, framerate, (1920,1200), isColor=False)
out2 = cv2.VideoWriter('D:\\videoCaptures\\output2.avi', fourcc, framerate, (1920,1200), isColor=False)

def init_ptp(camera, index):
    # PTP setup
    camera.PtpEnable.SetValue(False)
    camera.BslPtpPriority1.SetValue(128)
    camera.BslPtpProfile.SetValue("DelayRequestResponseDefaultProfile")
    camera.BslPtpNetworkMode.SetValue("Multicast")
    camera.BslPtpManagementEnable.SetValue(False)
    camera.BslPtpTwoStep.SetValue(False)
    camera.PtpEnable.SetValue(True)

def check_ptp(cameras):
    initialized = False
    locked = False

    # Wait until correctly initialized or timeout
    time1 = time.time()
    while not initialized:
        status_arr = np.zeros(cameras.GetSize(), dtype=np.bool_)
        for i, camera in enumerate(cameras):
            camera.PtpDataSetLatch.Execute()
            status_arr[i] = (camera.PtpStatus.GetValue() == 'Master' \
                            or camera.PtpStatus.GetValue() != 'Initializing')
        initialized = np.all(status_arr)
        if (time.time() - time1) > 3:
            if not initialized:
                warnings.warn('PTP not initialized -> Timeout')
            break

    # If correctly initialized, wait until settled or timeout
    if initialized:
        time2 = time.time()
        while not locked:
            status_arr = np.zeros(cameras.GetSize(), dtype=np.bool_)
            status_string = ''
            for i, camera in enumerate(cameras):
                camera.PtpDataSetLatch.Execute()
                status_arr[i] = (camera.PtpStatus.GetValue() == 'Master' \
                                or camera.PtpServoStatus.GetValue() == 'Locked')
                status_string += 'Camera {:d} locked: {} | '.format(i, status_arr[i])
            print(status_string)
            locked = np.all(status_arr)
[... 136 lines cut ...]
so PTP potentially is synchronized, when I check the frame output, the frames were actually not synchronized, but rather captured one after the other on the two cameras. However, with the same cameras and computer and code, when I tested again, PTP check process tell me it is synchronized, and frames are indeeded aligned (if I point both cameras to a milicsecond clock, and see the time on the frame). 

I am confused why with the same computer, same cameras, and same code, sometimes the PTP seem to successfully sync the cameras, while other times it does not; it seems that something is not stable in this system. If someone can share some ideas what could be the issue here, in the code, or in the setup, that will be extremely helpful and very apprecieated. In addition, I didn't find information on how to synchronously record from Pylon viewer only using the software; if this is possible and easy, please also help directing me to some instructions. Thank you!

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup & camera model(s) used

OS: Microsoft Windows 11 Enterprise
System Type: x64-based PC
RAM: 128 GB

POE interface card: IOI 4-CH PCIE 2.5GIGABIT POE+ CARD
cables: CAT6 ethernet cable
cameras: a2A1920-51gmPRO, ac21920-51gcBAS

### Runtime information:

```shell
python: 3.9.0 (default, Nov 15 2020, 08:30:55) [MSC v.1916 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 3.0.1 / 7.4.0.38864
```

canliu0414, 2024-03-08:
Following up on this, when I used 2 a2A1920-51gmPro cameras, instead of 1 a2A1920-51gmPro and 1 a2A1920-51gcBAS camera, the code seems to work more reliably; so it could be how I initiate/set up the cameras in my code, that is not optimized for certain camera?

SMA2016a, 2024-03-09:
I assume that that cameras are connected to 2 different ports.  in this case both cameras are working in PTP Master mode.

try to bind  both network ports  together using teaming. 

https://nocperu.com/en/component/content/article/13-news/89-nic-teaming-en-ms-windows-11

canliu0414, 2024-05-07:
I follow up on this thread as I got back to work on this. I am trying to get the exact timestamp that each frame is captured, and it seems like the timestamp feature in data chunk will work. However, I have problem interpretting the grab_result.BslChunkTimestampValue.Value value I get. I currently have 4 cameras connected through a POE board to the PC. It seems like only if I synchronize these cameras to a grand master clock (PTP hardware) then the time will be automatically synced to UTC time. If there is no grand master clock in my system (only 4 cameras competing to be the master), then the time value I get is number of ticks from the start of the camera?

Any suggestions on this? thank you!