basler/pypylon #732: Installation failed: "Wheel not supported" error when installing on 64bit Windows11

Aether59, 2024-03-10:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

## Summary
Goal is to install PyPylon on a 64-bit Windows 11 machine.   With Python3, pip, and Pylon installed, installation fails on "ERROR: pypylon-3.0.1-cp311-cp311-win_amd64.whl is not a supported wheel on this platform." 

Details below. Thanks in advance for any help.

## System
OS:  Windows 11 Pro
Processor	Intel(R) Core(TM) i3-1005G1 CPU @ 1.20GHz   1.20 GHz
Installed RAM	8.00 GB (7.75 GB usable)
System type	64-bit operating system, x64-based processor

## Dependencies
- Pylon installed from file 'basler_pylon_7_4_0_14900'.  Viewer loads ok.
- C:\Users\User\OneDrive\Desktop>python --version
Python 3.12.2
- C:\Users\User\OneDrive\Desktop>pip3 --version
pip 24.0 from C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.12_3.12.752.0_x64__qbz5n2kfra8p0\Lib\site-packages\pip (python 3.12)

## Downloaded pypylon-3.0.1-cp311-cp311-win_amd64.whl to local directory
Directory of C:\Users\User\OneDrive\Desktop
...
10/03/2024  08:18       145,682,040 pypylon-3.0.1-cp311-cp311-win_amd64.whl

## Install fails
C:\Users\User\OneDrive\Desktop>pip3 install pypylon-3.0.1-cp311-cp311-win_amd64.whl
Defaulting to user installation because normal site-packages is not writeable
ERROR: pypylon-3.0.1-cp311-cp311-win_amd64.whl is not a supported wheel on this platform.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

See above

### Runtime information:

```shell
See above
```

HighImp, 2024-03-10:
Hi, 

Pypylon is not jet released for python 3.12
Please install python 3.11 and try again.

The Release for 3.12 will be in Mai 2024, according to #709.

Best regards

Aether59, 2024-03-10:
Yes, that did it. Thanks very much.

closed 2024-03-10 by Aether59