basler/pypylon #734: Line Scan Camera not detected in Pypylon

camron-musashi, 2024-03-15:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

My goal is to use the Basler Racer raL4096-24gm camera to get a line scan of a part, I am able to open and configure settings in Pylon Viewer but when I try to make a camera object in python nothing is detected.

```from pypylon import pylon
from pypylon import genicam

import sys

# Number of images to be grabbed.
countOfImagesToGrab = 100

# The exit code of the sample application.
exitCode = 0

try:
    # Create an instant camera object with the camera device found first.
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    camera.Open()

    # Print the model name of the camera.
    print("Using device ", camera.GetDeviceInfo().GetModelName())

    # demonstrate some feature access
    new_width = camera.Width.Value - camera.Width.Inc
    if new_width >= camera.Width.Min:
        camera.Width.Value = new_width

    # The parameter MaxNumBuffer can be used to control the count of buffers
    # allocated for grabbing. The default value of this parameter is 10.
    camera.MaxNumBuffer.Value = 5

    # Start the grabbing of c_countOfImagesToGrab images.
    # The camera device is parameterized with a default configuration which
    # sets up free-running continuous acquisition.
    camera.StartGrabbingMax(countOfImagesToGrab)

    # Camera.StopGrabbing() is called automatically by the RetrieveResult() method
    # when c_countOfImagesToGrab images have been retrieved.
    while camera.IsGrabbing():
        # Wait for an image and then retrieve it. A timeout of 5000 ms is used.
        grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)

        # Image grabbed successfully?
        if grabResult.GrabSucceeded():
            # Access the image data.
            print("SizeX: ", grabResult.Width)
            print("SizeY: ", grabResult.Height)
            img = grabResult.Array
            print("Gray value of first pixel: ", img[0, 0])
        else:
            print("Error: ", grabResult.ErrorCode, grabResult.ErrorDescription)
        grabResult.Release()
    camera.Close()

except genicam.GenericException as e:
    # Error handling.
    print("An exception occurred.")
    print(e)
    exitCode = 1

sys.exit(exitCode)
```
Error Code: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'tlfactory.cpp', line 694)    

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

x64 Architechture 
Windows 11 Operating System
Camera: Basler raL4096-24gm

### Runtime information:

```shell
python: 3.9.13 (tags/v3.9.13:6de2ca5, May 17 2022, 16:36:42) [MSC v.1929 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 3.0.1 / 7.4.0.38864
```

SMA2016a, 2024-03-15:
Would post the IP address and subnetmask of the camera and network card.

camron-musashi, 2024-03-19:
Camera:
IP: 192:168:2.21:3956
Subnet mask: 255.255.255.0

Network Card:
IPv4: 192.168.1.99
Subnet Mask: 255.255.0.0

SMA2016a, 2024-03-20:
The camera must be findable using unicast communication. 
The Subnet not mating each other.

Camera:
IP: 192:168:1.21:3956
Subnet mask: 255.255.255.0

Network Card:
IPv4: 192.168.1.99
Subnet Mask: 255.255.255.0

or

Camera:
IP: 192:168:2.21:3956
Subnet mask: 255.255.0.0

Network Card:
IPv4: 192.168.1.99
Subnet Mask: 255.255.0.0

camron-musashi, 2024-03-20:
Changed camera's Subnet mask to match computers, and now connects. However image grab is not working. 

```
ip_address = '192.168.2.21:3956'
subnet_mask = '255.255.0.0'

factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()

empty_camera_info.SetPropertyValue('IpAddress', ip_address)
empty_camera_info.SetPropertyValue('SubnetMask', subnet_mask)

camera_device = factory.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)
camera.Open()
print("Using device ", camera.GetDeviceInfo().GetModelName())
camera.Height.SetValue(height)

camera.TriggerSelector.Value = 'FrameStart'
camera.TriggerMode.Value = 'On'
camera.TriggerSource.Value = 'Software'

camera.StartGrabbingMax(countOfImagesToGrab)
camera.TriggerSoftware.Execute()
print('Camera triggered')
print("starting capture")

actuatorThread = myThread("Actuator-thread", 1)
actuatorThread.start()

#don't grab any images until enought time has passsed for the first frame of the line scan to come through
#convert exposure from us to s
timeDelay = (exp / 1000000) * height
time.sleep(timeDelay)

while camera.IsGrabbing():
    print('Grabbing')
    result = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
    if result.GrabSucceeded():
        image = result.Array
        result.Release()
        path = "C:\\Users\\camron.sabahi-pourka\\Desktop"
        cv2.imwrite(path + "\\light_1.bmp", image)
        print("Image saved successfully.")
        break
    if not result.GrabSucceeded():
        print('Grab failed')
    result.Release()
    ```

camron-musashi, 2024-03-20:
The grabs were all failing/time out-ing/crashing my computer.  

> Type	Time	Source	Message
Error	2024-03-20 13:31:26.314	Basler raL4096-24gm (23861554)	Image acquisition on "Basler raL4096-24gm (23861554)" failed! Error: "The buffer was incompletely grabbed. This can be caused by performance problems of the network hardware used, i.e., the network adapter, switch, or Ethernet cable. Buffer underruns can also cause image loss. To fix this, use the pylonGigEConfigurator tool to optimize your setup and use more buffers for grabbing in your application to prevent buffer underruns."

This is what I would get when I try to capture a frame on Pylon Viewer. Very weird because I wasn't getting these errors in the previous days I was using the camera

HighImp, 2024-03-21:
Have you tried the GigE-Configurator in the Pylon Viewer?

camron-musashi, 2024-03-21:
Yes. However, when we would click the configuration button it would just close the window

SMA2016a, 2024-04-03:
Does your script work, if you do not write the image to hardware?

camron-musashi, 2024-04-04:
Yes the script works, and I'm able to connect to the camera however now I get this error
`_genicam.TimeoutException: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped. : TimeoutException thrown (file 'instantcameraimpl.h', line 1036)`

This is my code

```factory = pylon.TlFactory.GetInstance()
ptl = factory.CreateTl('BaslerGigE')
empty_camera_info = ptl.CreateDeviceInfo()

empty_camera_info.SetPropertyValue('IpAddress', ip_address)
empty_camera_info.SetPropertyValue('SubnetMask', subnet_mask)

camera_device = factory.CreateDevice(empty_camera_info)
camera = pylon.InstantCamera(camera_device)

#Camera Setup for capturing
countOfImagesToGrab = 10
camera.Open()
print("Using device ", camera.GetDeviceInfo().GetModelName())
camera.Height.SetValue(height)
camera.PixelFormat.Value = "Mono8"
camera.TriggerSelector.Value = 'FrameStart'
camera.TriggerMode.Value = 'On'
camera.TriggerSource.Value = 'Software'
# camera.MaxNumBuffer.Value = 5
# camera.AcquisiutionMode.Value = 'Single Frame'
# camera.AcquisitionStart.Execute()

camera.StartGrabbingMax(countOfImagesToGrab)
camera.TriggerSoftware.Execute()
print('Camera triggered')
print("starting capture")

actuatorThread = myThread("Actuator-thread", 1)
actuatorThread.start()

#don't grab any images until enought time has passsed for the first frame of the line scan to come through

timeDelay = (exp / 1000000) * height
time.sleep(timeDelay - 1.5)
degPerS = velocity / (16*14) * 1.8
imagingTime = int(360 // degPerS)
print(imagingTime)

while camera.IsGrabbing():
    print('Grabbing')
    result = camera.RetrieveResult(6000, pylon.TimeoutHandling_ThrowException)
    if result.GrabSucceeded():
        image = result.Array
        result.Release()
        path = "C:\\Users\\camron.sabahi-pourka\\Desktop"
        cv2.imwrite(path + "\\light_2.bmp", image)
        print("Image saved successfully.")
        result.Release()
        camera.StopGrabbing()
        camera.Close()
        break
    if not result.GrabSucceeded():
        print('Grab failed')
        result.Release()
        camera.StopGrabbing()
        camera.Close()
        print('Camera closed and grabbing stopped')
    result.Release()

result.Release()
camera.StopGrabbing()
camera.Close()```

SMA2016a, 2024-04-05:
remove this line
ptl = factory.CreateTl('BaslerGigE')

try this:

```
ip_address = '192.168.1.40'
tlf = pylon.TlFactory.GetInstance()
cam_info = pylon.DeviceInfo()   
cam_info.SetIpAddress(ip_address)
cam = pylon.InstantCamera(tlf.CreateDevice(cam_info))
```

SMA2016a, 2024-04-10:
@camron-musashi  did you try this?

camron-musashi, 2024-04-10:
Yes I did. For some reason I have no issues one on computer, however on the other computer (exact same computers) it throws errors and often crashes the computer. The computer shows issues both with pypylon and the Pylon Viewer GUI, but it was installed to have the packages needed for USB and gigE.

kesaroid, 2024-05-10:
this bug 
```
_genicam.TimeoutException: Grab timed out. Possible reasons are: The image transport from the camera device is not working properly, e.g., all GigE network packets for streaming are dropped; The camera uses explicit triggering (see TriggerSelector for more information) and has not been triggered; Single frame acquisition mode is used and one frame has already been acquired; The acquisition has not been started or has been stopped. : TimeoutException thrown (file 'instantcameraimpl.h', line 1036)```
seems to related to the camera disconnection issue that you may also be seeing. Do the cameras work fine through pylon-viewer?