basler/pypylon #736: Function 'PylonCreateDeviceByIndex' failed in source file ,device cannot be operated on an USB 2.0

panguoxian, 2024-03-18:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Does the ML500-75M camera not support USB2.0? My embedded board only has a USB2.0 interface.
Works fine on X86X64 host.
Does ML500-75M only work with USB 3.0 hardware interface?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

CPU architecture: ARM64
Operating System: Linux

Cameras: ML500-75M

### Runtime information:

```shell
**Plug in camera：**

[ 3261.839330] usb 2-1: new high-speed USB device number 2 using ehci-platform
[ 3261.989041] usb 2-1: New USB device found, idVendor=2676, idProduct=ba02, bcdDevice= 0.00
[ 3261.989220] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3261.989258] usb 2-1: Product: ML500-75M
[ 3261.989286] usb 2-1: Manufacturer: Basler
[ 3261.989311] usb 2-1: SerialNumber: 22190150

**Running SimpleGrab compiled in C：**

PylonC error #c2000001 'Failed to create the device.' (0xc2000001).
Function 'PylonCreateDeviceByIndex' failed in source file '../../../../../Pylon/PylonC/PylonC/PylonC.cpp' line #1836
Caused by an exception thrown from source file '../../../../../Pylon/PylonTL_USB/PylonUSB/PylonUsbTL.cpp' line #497.
Failed to open device '2676:ba02:2:1:2' for XML file download. Error: 'The device cannot be operated on an USB 2.0 port. The device requires an USB 3.0 compatible port.'
```

SMA2016a, 2024-03-18:
no.  You can use a dart camera at USB 2 port.

panguoxian, 2024-03-19:
But prompt：
The device cannot be operated on an USB 2.0 port. The device requires an USB 3.0 compatible port

What causes?
The USB3.0 interface works normally. The above prompt appears on the USB2.0 interface....

panguoxian, 2024-03-19:
> no. You can use a dart camera at USB 2 port.

But prompt：
The device cannot be operated on an USB 2.0 port. The device requires an USB 3.0 compatible port

What causes?
The USB3.0 interface works normally. The above prompt appears on the USB2.0 interface....

closed 2024-03-19 by panguoxian