basler/pypylon #737: in application some cameras run some of the doesnt run

vseyok1, 2024-03-25:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

We are trying to use threading. However, in our application some of cameras can capture, and the other ones gives grabsuccedded false. When we are running in serial it works but it takes too much time. We are using 8 cameras.

    def take_photo(self):

        self.updateIpList()
        self.org_image_list = {}
        if self.video_counter % 2 != 0:
            self.start_video()

        #temp_ip_list = ['192.168.1.1', '192.168.1.2']
        gain_list = []
        exp_list = []
        for le in self.lineEdit_list:
            gain_list.append(le[0].text())
            exp_list.append(le[1].text())
        with open("cam_ip.json", 'r') as cam_ip_file:
            cam_ips = json.load(cam_ip_file)
        self.number_list = []

        for i, ke in enumerate(cam_ips):
            if cam_ips[ke]:
                self.number_list.append(i)
        print("numberlist", self.number_list)

        tlFactory = pylon.TlFactory.GetInstance()

    # Get all attached devices and exit application if no device is found.
        devices = tlFactory.EnumerateDevices()
        #print(devices)
        if len(devices) == 0:
            raise pylon.RuntimeException("No camera present.")

        # Create an array of instant cameras for the found devices and avoid exceeding a maximum number of devices.
        cameras = pylon.InstantCameraArray(min(len(devices), 8))
        #print(cameras)
        cams = []
        # Create and attach all Pylon Devices.
        #print("beforeip",ip_address)
        for i in range(len(self.ip_list)):
            for idx, cam in enumerate(cameras):
                cam.Attach(tlFactory.CreateDevice(devices[idx]))
                if cam.GetDeviceInfo().GetIpAddress() == self.ip_list[i]:
                    cams.append(cam)
                    break
            # setup named window per camera

        import queue
        image_queue = queue.Queue()

        def capture_image(i, camera):
            acq=False
            camera.Open()
            camera.ExposureTimeAbs = int(exp_list[i])
            camera.GainRaw.SetValue(int(gain_list[i]))
            #pylonda çeklilen fotonun opencv ye uymas için
            converter = pylon.ImageFormatConverter()
            converter.OutputPixelFormat = pylon.PixelType_BGR8packed
            converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

            camera.StartGrabbing(1)#pylon.GrabStrategy_LatestImageOnly

            grab = camera.RetrieveResult(2000, pylon.TimeoutHandling_Return)
         
            if grab.GrabSucceeded():
                #fotoğraf çekme başarılı ise
                image = converter.Convert(grab)
                img = image.GetArray()
                img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
                flip = cv2.flip(img, 1)
                image1 = cv2.cvtColor(img, cv2.COLOR_RGB2BGR)
                trueImageFolder = 'C:/temp/'

                print(trueImageFolder + 'resim-' + str(i) + '.png')
                isWritten = cv2.imwrite(trueImageFolder+'resim-' + str(i) + '.png', image1)
                print("kaydedld" + str(isWritten))

                convert_qt = QImage(img.data, img.shape[1], img.shape[0], QImage.Format_RGB888)
                img = convert_qt.scaled(flip.shape[1], flip.shape[0], Qt.KeepAspectRatio)
                acq=True
                print(4)
                image_queue.put((i, convert_qt, acq))
            else:
                img = []
                print("the image is failed to acquire!")
            cams[i].Close()

            return convert_qt,acq

        threads = []

        for i, camera in enumerate(cameras):
            thread = threading.Thread(target=capture_image, args=(self.number_list[i], camera))
            threads.append(thread)

        for thread in threads:
            thread.start()

        for thread in threads:
            thread.join()

        while not image_queue.empty():
            i, img, acq = image_queue.get()
            print(acq)
            im = QtGui.QPixmap(img)
            print(6)
            self.label_list[i].setPixmap(im)
            print(7)
            self.org_image_list[i]=img

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

basler ace gige cameras
Intel(R) Core(TM) i7-10700F CPU @ 2.90GHz   2.90 GHz
8 gb ram

### Runtime information:

```shell
python: 3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 1.7.2 / 6.1.1.18345
```

SMA2016a, 2024-03-27:
Most properly, this is an issue related to bandwidth. Maybe your network devices are not able to manage the burst while both cameras are transfer the image at the same time.

1) Try to set the camera parameter interpacket delay. 
2) set the packet size to 9000 if your NIC supports the bigger MTU size.

vseyok1, 2024-04-02:
After I optimize the packet size with using bandwidth optimizer(around 2400), it works. Thank you!

SMA2016a, 2024-04-03:
thanks for the update

closed 2024-04-03 by SMA2016a