basler/pypylon #738: Another memory leak when using BaslerCompressionBeyond (this time when sending grab array to differnet thread)

hlacikd, 2024-03-29:
### Describe the issue:

When using BaslerCompressionBeyond and doing decompression within grabbing loop

```
    while camera.IsGrabbing():
        # Wait for an image and then retrieve it. A timeout of 5s is used.
        with camera.RetrieveResult(5 * 1000, pylon.TimeoutHandling_Return) as grab:
            # Image grabbed successfully?
            if grab.GrabSucceeded():
                try:
                    # Add frame to redis
                    f(decompressor.DecompressImage(grab).Array if decompressor else grab.Array)
                    tic += 1
                except pylon.GenericException as e:
                    logger.debug(f"{key}> {e}")
            else:
                logger.debug(f"{key}> {grab.GetErrorDescription()}")
                continue
```
and sending frame array to different thread (using ThreadPoolExecutor) -- so grabbing loop is not blocked (in my case on storring image to redis) Memory Leak is happening -- aka memory is increasing after each iteration through grabbing loop.

Not sure why it is happening since I am sending allready grabbed array and not buffer or whole grab object. Please see complete code attached.

When I disable threadpool i can no longer see Memory Leak.

Also --- this does not happen in case BaslerCompressionBeyond is not used. Aka posting frame array to different thread.
So it must be related to pylon.ImageDecompressor() or compression descriptor somehow?

### Reproduce the code example:

```python
from argparse import ArgumentParser
from concurrent.futures.thread import ThreadPoolExecutor
from dataclasses import dataclass, field
from pickle import dumps
from time import time

from loguru import logger
from redis import ConnectionPool, Redis
from redis.exceptions import RedisError

from pypylon import pylon

def get_args():
    parser = ArgumentParser("Basler camera driver")

    parser.add_argument("--redis_host", type=str, default="redis.db", help="Hostname for redis")
    parser.add_argument("--redis_key", type=str, default="basler", help="Key for redis")

    parser.add_argument("--model", type=str, default=False, help="Camera model")
    parser.add_argument("--address", type=str, default=False, help="Camera IP address")
    parser.add_argument("--pfs", type=str, default="/app/pfs/basler.pfs", help="Camera nodemap file")
    parser.add_argument("--fps", type=int, default=30, help="Basler FPS")

    parser.add_argument("--rotate", type=lambda x: x.lower() in ["true", "1"], default=False, help="Rotate camera")

    return parser.parse_args()

FLAGS = get_args()
TIC = 60

@dataclass(order=True)
class Camera:
    # device
    device: pylon.DeviceInfo
    config: str
    # info
    model: str = field(init=False)
    sn: int = field(init=False)
[... 182 lines cut ...]
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Basler a2A5320-23umPRO (USB3) on Ubuntu 22.04 (arn64) with pypylon==3.0.1

### Runtime information:

```shell
root@basler-78978b9445-6fnvc:/app# python3
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys, pypylon.pylon, platform
>>> print(f'python: {sys.version}')
python: 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
>>> print(f'platform: {sys.platform}/{platform.machine()}/{platform.release()}')
platform: linux/aarch64/5.10.160-legacy-rk35xx
>>> print(f'pypylon: {pypylon.pylon.__version__} / {".".join([str(i) for i in pypylon.pylon.GetPylonVersion()])}')
pypylon: 3.0.1 / 7.4.0.38864
>>>
```

SMA2016a, 2024-03-31:
Thanks for your well-prepared report. We will check it and get back to you soon with the result

hlacikd, 2024-04-02:
> currently can't reproduce your issue. can you install heaptrack and run your code inside of heaptrack?
> 
> `apt-get install heaptrack-gui`
> 
> `heaptrack python sample_script.py`
> 
> to visualize `heaptrack-gui` and open the saved heaptrack file.
> 
> this should give you a deeper insight into what is leaking.

hello @thiesmoeller . thank you for response i will try to provide it to you.

I forgot to mention that i am using python library redis[hiredis] , (pip install redis[hiredis]) which uses C library hiredis and is faster than pure python redis implementation.

And i was able to investigate to that that it happens only with redis[hiredis]. When i use only pip install redis (pure python implementation of redis it is not happening).
However i am using redis[hiredis] in many other projects and that memory leak happens only in this case.

Thank you for spending your time with this.

thiesmoeller, 2024-04-02:
For the testing on our side we replaced the redis calls by just pushing into a short deque.

To split the possible root cause, can you replace adding the pypylon grab.Array to redis by generating a simple numpy empty array of same size?

hlacikd, 2024-04-03:
yes as i said, when i do not use basler compression beyond my grabbing loop is pushing just grab.Array

```python
 with camera.RetrieveResult(5 * 1000, pylon.TimeoutHandling_Return) as grab:
            # image grabbed successfully?
            if grab.GrabSucceeded():
                try:
                    # add frame to redis
                    **f(decompressor.DecompressImage(grab).Array if decompressor else grab.Array)**
                    tic += 1
                except pylon.GenericException as e:
                    logger.debug(f"{key}> {e}")
            else:
                logger.debug(f"{key}> {grab.GetErrorDescription()}")
                continue
```

f(grab.Array is called)

f  self.pool.submit(self._add_frame, frame) in Frames call, which submits grab.Array to _add_frame

    def _add_frame(self, frame, maxlen=30, expiry=5):
        try:
            # add frame to redis
            with self.db.pipeline() as pipe:
                pipe.xadd(self.key, {"frame": dumps(frame), "timestamp": time(), "format": self.format}, maxlen=maxlen)
                pipe.expire(self.key, expiry)
                pipe.execute()
        except RedisError as e:
            logger.error(f"{self.key}> {e}")

And this works without memory leak!

However when compression beyond is enabled as you can see my grabbing loop does

```python
decompressor.DecompressImage(grab).Array
```
so decompressed image is put to f()

and this causes Memory Leak which happens very fast on 5Mpix camera. (16GB within 5minutes)

of course i can replace grab.Array with random numpy image, so my grabbing loop looks like this

```python
while camera.IsGrabbing():
        # wait for an image and then retrieve it. A timeout of 5s is used.
        with camera.RetrieveResult(5 * 1000, pylon.TimeoutHandling_Return) as grab:
            # image grabbed successfully?
            if grab.GrabSucceeded():
                try:
                    # add frame to redis
                    # f(decompressor.DecompressImage(grab).Array if decompressor else grab.Array)
                    # generate random numpy array
                    img = np.random.randint(low=0, high=256, size=(2200, 2200), dtype=np.uint8)
                    f(img)
                    tic += 1
                except pylon.GenericException as e:
                    logger.debug(f"{key}> {e}")
            else:
                logger.debug(f"{key}> {grab.GetErrorDescription()}")
                continue
```

and it works (no memory leak)

To sum up

When using redis[hiredis]==5.0.3 and pypylon==3.0.1 and ThreadPoolExecutor

1) Works for grab.Array (when compression is disabled)
2) Causes memory leak when image is decompressed 
3) Works for random np images

hlacikd, 2024-04-03:
@thiesmoeller  do you have possibility to just run my script (code example) it is really easy to run it 

just pass --model  and --pfs for USB3 camera , for example

python3 basler.py --model=a2A5320-23umPRO --pfs=basler.pfs 

you can run redis via docker 

`docker run --net=host redis`

this will bind it to your localhost and specify host address via --redis_host=localhost

thiesmoeller, 2024-04-03:
As a first quick test tried to reproduce on x86_64 using your configuration.

Can't reproduce the memleak currently

Can you cross-check on x86_64 with the same versions?

Used the conda build of python 3.10.12.

or if you don't have heaptrack on your arm64 system, do a quick check with valgrind:

`PYTHONMALLOC=malloc valgrind python memleak_script.py`

This was my check. redis is installed with hiredis addon
```
2024-04-03 16:42:15.508 | INFO     | __main__:_pre_init:58 - Added camera model: a2A4508-6gcPRO type: BaslerGigE sn: 40271242
2024-04-03 16:42:16.088 | INFO     | __main__:_init:66 - Initialized camera model: a2A4508-6gcPRO type: BaslerGigE sn: 40271242
2024-04-03 16:42:16.128 | INFO     | __main__:_set_defaults:72 - Loaded default settings for model: a2A4508-6gcPRO type: BaslerGigE sn: 40271242
2024-04-03 16:42:17.790 | INFO     | __main__:_setup:78 - Loaded custom settings for model: a2A4508-6gcPRO type: BaslerGigE sn: 40271242
2024-04-03 16:42:17.793 | INFO     | __main__:_post_init:86 - FPS for model: a2A4508-6gcPRO type: BaslerGigE sn: 40271242 is 30
2024-04-03 16:42:17.797 | INFO     | __main__:_post_init:94 - Pixel format for model: a2A4508-6gcPRO type: BaslerGigE sn: 40271242 is BayerGB8
2024-04-03 16:42:17.804 | INFO     | __main__:_post_init:99 - Compression for model: a2A4508-6gcPRO type: BaslerGigE sn: 40271242 is enabled
2024-04-03 16:42:18.001 | INFO     | __main__:grab_loop:162 - Grabbing basler:0
2024-04-03 16:42:26.195 | INFO     | __main__:grab_loop:196 - basler:0 > fps (app/camera): 7.37 / 7.42, payload: 13.21 MB, compression ratio: 72.80 %, size: 12.84 MB
2024-04-03 16:42:26.195 | DEBUG    | __main__:grab_loop:197 - basler:0 > temperature (current/max): 58.12 / 59.00 °C, status: Ok
2
```

hlacikd, 2024-04-04:
this is valgrind output , is it any help to you? (on that arm64v8 device)

```
root@basler-699b9d86bf-gdt54:/app# PYTHONMALLOC=malloc valgrind python3 basler.py --fps=0 --model=a2A5320-23umPRO
==733== Memcheck, a memory error detector
==733== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==733== Using Valgrind-3.18.1 and LibVEX; rerun with -h for copyright info
==733== Command: python3 basler.py --fps=0 --model=a2A5320-23umPRO
==733== 
--733-- Warning: DWARF2 CFI reader: unhandled DW_OP_ opcode 0x92

valgrind: m_debuginfo/readdwarf.c:2761 (copy_convert_CfiExpr_tree): Assertion 'srcix >= 0 && srcix < VG_(sizeXA)(srcxa)' failed.

host stacktrace:
==733==    at 0x58040FEC: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x58041143: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x580412AB: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x580C1617: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x580C17B3: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x580C6DC3: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x5807A887: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x5806FA37: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x5809EB37: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x580AAF43: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x5809AA63: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x5809636B: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x580987BF: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0x580DEC8F: ??? (in /usr/libexec/valgrind/memcheck-arm64-linux)
==733==    by 0xFFFFFFFFFFFFFFFF: ???

sched status:
  running_tid=1

Thread 1: status = VgTs_Runnable syscall 222 (lwpid 733)
==733==    at 0x401D7AC: __mmap64 (mmap64.c:58)
==733==    by 0x401D7AC: mmap (mmap64.c:46)
==733==    by 0x40079CB: _dl_map_segments (dl-map-segments.h:140)
==733==    by 0x40079CB: _dl_map_object_from_fd (dl-load.c:1258)
==733==    by 0x4008ACF: _dl_map_object (dl-load.c:2327)
==733==    by 0x4002E6F: openaux (dl-deps.c:64)
==733==    by 0x4ACD307: _dl_catch_exception (dl-error-skeleton.c:208)
==733==    by 0x40033C3: _dl_map_object_deps (dl-deps.c:248)
==733==    by 0x400C77B: dl_open_worker_begin (dl-open.c:592)
==733==    by 0x4ACD307: _dl_catch_exception (dl-error-skeleton.c:208)
==733==    by 0x400BEF7: dl_open_worker (dl-open.c:782)
==733==    by 0x4ACD307: _dl_catch_exception (dl-error-skeleton.c:208)
==733==    by 0x400C2FB: _dl_open (dl-open.c:883)
==733==    by 0x4A196E3: dlopen_doit (dlopen.c:56)
==733==    by 0x4ACD307: _dl_catch_exception (dl-error-skeleton.c:208)
==733==    by 0x4ACD3CF: _dl_catch_error (dl-error-skeleton.c:227)
==733==    by 0x4A191BF: _dlerror_run (dlerror.c:138)
==733==    by 0x4A19783: dlopen_implementation (dlopen.c:71)
==733==    by 0x4A19783: dlopen@@GLIBC_2.34 (dlopen.c:81)
==733==    by 0x325F1F: ??? (in /usr/bin/python3.10)
==733==    by 0x324333: ??? (in /usr/bin/python3.10)
==733==    by 0x20D59B: ??? (in /usr/bin/python3.10)
==733==    by 0x1F6D8F: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x20D347: _PyFunction_Vectorcall (in /usr/bin/python3.10)
==733==    by 0x1F977B: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x20D347: _PyFunction_Vectorcall (in /usr/bin/python3.10)
==733==    by 0x1F53F7: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x20D347: _PyFunction_Vectorcall (in /usr/bin/python3.10)
==733==    by 0x1F52BB: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x20D347: _PyFunction_Vectorcall (in /usr/bin/python3.10)
==733==    by 0x1F52BB: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x20D347: _PyFunction_Vectorcall (in /usr/bin/python3.10)
==733==    by 0x1F52BB: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x20D347: _PyFunction_Vectorcall (in /usr/bin/python3.10)
==733==    by 0x20C26F: ??? (in /usr/bin/python3.10)
==733==    by 0x2F8EE7: _PyObject_CallMethodIdObjArgs (in /usr/bin/python3.10)
[... 57 lines cut ...]
==733==    by 0x1F77AB: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x2F175F: ??? (in /usr/bin/python3.10)
==733==    by 0x2F15E3: PyEval_EvalCode (in /usr/bin/python3.10)
==733==    by 0x2F986B: ??? (in /usr/bin/python3.10)
==733==    by 0x20D59B: ??? (in /usr/bin/python3.10)
==733==    by 0x1F6D8F: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x20D347: _PyFunction_Vectorcall (in /usr/bin/python3.10)
==733==    by 0x1F977B: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
==733==    by 0x20D347: _PyFunction_Vectorcall (in /usr/bin/python3.10)
==733==    by 0x1F53F7: _PyEval_EvalFrameDefault (in /usr/bin/python3.10)
client stack range: [0x1FFEFEE000 0x1FFF000FFF] client SP: 0x1FFEFF5E40
valgrind stack range: [0x1002DBC000 0x1002EBBFFF] top usage: 19824 of 1048576

Note: see also the FAQ in the source distribution.
It contains workarounds to several common problems.
In particular, if Valgrind aborted or crashed after
identifying problems in your program, there's a good chance
that fixing those problems will prevent Valgrind aborting or
crashing, especially if it happened in m_mallocfree.c.

If that doesn't help, please report this bug to: www.valgrind.org

In the bug report, send all the above text, the valgrind
version, and what OS and version you are using.  Thanks.
```

hlacikd, 2024-04-04:
> 
> Can you cross-check on x86_64 with the same versions?
> 
> Used the conda build of python 3.10.12.
> 

@thiesmoeller  just tested in on my computer (x86_64) and i can confirm that memory overflow is present only on arm64v8 architecture. Do you have access to arm64v8 board? Or how can I be a help more so you can debug it?

this is my machine info i have tested it on 

```
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys, pypylon.pylon, platform
: {sys.version}')
print(f'platform: {sys.platform}/{platform.machine()}/{platform.release()}')
print(f'pypylon: {pypylon.pylon.__version__} / {".".join([str(i) for i in pypylon.pylon.GetPylonVersion()])}')>>> print(f'python: {sys.version}')
python: 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
>>> print(f'platform: {sys.platform}/{platform.machine()}/{platform.release()}')
platform: linux/x86_64/6.5.0-26-generic
>>> print(f'pypylon: {pypylon.pylon.__version__} / {".".join([str(i) for i in pypylon.pylon.GetPylonVersion()])}')
pypylon: 3.0.1 / 7.4.0.38864
>>> 
```

thiesmoeller, 2024-04-05:
This is really special..

x86_64 both can't reproduce
arm64 (native) ubuntu 20.04 ( pyenv python 3.10 ) arm64 xavier agx can't reproduce...
arm64 (docker) ubuntu 22.04 python 3.10  arm64 xavier agx can't reproduce...

can you check in a docker environment on your embedded system?:

Dockerfile is
```dockerfile
FROM ubuntu:22.04

COPY pylon_7_4_0_14900_linux_aarch64_debs.tar.gz /tmp

RUN tar -xvf /tmp/pylon_7_4_0_14900_linux_aarch64_debs.tar.gz
# Set environment variables to avoid interactive dialog while installing packages
ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update

RUN apt-get install -y  ./*.deb
RUN apt-get install -y python3 python3-pip
RUN pip install pypylon loguru numpy redis[hiredis]

COPY issue_738.py /tmp
COPY basler.pfs /tmp
```

`docker build -f Dockerfile_issue_738  -t issue_738 . `

To build the container you have to put the pylon 7.4 installer tar alongside the dockerfile.
Also your script from first entry and the pfs file from last comment.

The pylon installer is not required to run pypylon, 
but if you can reproduce on your side it would be good to rebuild pypylon locally.

start redis:

`docker run -d --net=host redis`

start test container

`docker run --net=host --rm -ti issue_738`

in container I call:
`python3 issue_738.py --pfs basler.pfs --redis_host localhost`

hlacikd, 2024-04-05:
@thiesmoeller  thanks for this. i am going to try. I am actually running it in dockerized container myself, actually it is deployed on k3s (edge kubernetes). I will get back to you and many thanks!

hlacikd, 2024-04-05:
I have used the following dockerfile (same as you except downloaded tar.gz directly

```
FROM ubuntu:22.04
WORKDIR /app

ADD https://www2.baslerweb.com/media/downloads/software/pylon_software/pylon_7_4_0_14900_linux_aarch64_debs.tar.gz /tmp

RUN tar -xvf /tmp/pylon_7_4_0_14900_linux_aarch64_debs.tar.gz
# Set environment variables to avoid interactive dialog while installing packages
ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update

RUN apt-get install -y  ./*.deb
RUN apt-get install -y python3 python3-pip
RUN pip install pypylon loguru numpy redis[hiredis]

COPY  basler.py .
ENTRYPOINT ["python3", "/app/basler.py"]
```

and i can confirm memory leak is present :)

I am going to do the following.

1. run it outside of k3s kubernetes directly on host (ubuntu 22.04 arm64v8)
3. i will try different camera than a2A5320-23umPRO, i have a2A1920 .. would be interesting if it was specific to this camera model.

and i will get back to you, have a nice weekend.

hlacikd, 2024-04-06:
Hello, honestly i gave up on this one. I really do suspected kubernetes has something with this but then i launched it on host machine itself and memory leak was there as well.
[image]

I end up with a simple trick where i serialize grab array to bytes using pickle.dumps and sending it to another thread already as bytes (i was doing that anyway inside _add_frame function on adding frame to redis. 

So i ended up with this inside grabbing loop
```
            # image grabbed successfully?
            if grab.GrabSucceeded():
                try:
                    # add frame to redis
                    frame = decompressor.DecompressImage(grab).Array if decompressor else grab.Array
                    f(dumps(frame))
                    tic += 1
                except pylon.GenericException as e:
                    logger.debug(f"{key}> {e}")
```

and _add_frame has dumps removed

```
    def _add_frame(self, frame, maxlen=10, expiry=5):
        try:
            # add frame to redis
            with self.db.pipeline() as pipe:
                pipe.xadd(self.key, {"frame": frame, "timestamp": time(), "format": self.format}, maxlen=maxlen)
                pipe.expire(self.key, expiry)
                pipe.execute()
        except RedisError as e:
            logger.error(f"{self.key}> {e}")
```

and it ended up working without memory leak. So i guess its good enough for me.

I did a debuging as well , and found out that `decompressor.DecompressImage(grab).Array` produces just normal array , of type uint8 so i have no idea what could be wrong with that.

What really messes with my head is why it works when basler beyond compression is not enabled (since in both cases it end ups as same array of same datetype, but i also believe that since you have not reproduced this bug it will be related to software libraries on my side (even though we used docker to isolate enviroment).

I suggest to close this as non reproducible if you have no ideas , mb someone will stumbe upon this in future and will be able to provide better insight.

Anyone thank you for your help, much appreciated.

thiesmoeller, 2024-04-06:
What is your exact host machine ? Exact host OS Version, kernel version 

Are you using exactly the small PFS file you mention?

Have you tested on another machine with fresh setup

We have to make our setups identical.

hlacikd, 2024-04-08:
> Even if we set this to nom reproducible ..
> 
> What is your exact host machine ? Exact host OS Version, kernel version

its Armbian 23.11 on arm64, kernel is Linux 5.10.160

@thiesmoeller is there a way to contact you privately (via email?) i can share with you more details which i cannot share in public.

Thanks

hlacikd, 2024-04-09:
unable to reproduce on different hardware as well, i will live with workaround for now. if i will have more info i will try to provide it here. thanks

closed 2024-04-09 by hlacikd