basler/pypylon #740: BUG: Segmentation fault at conda

kkuranoCCS, 2024-04-01:
### Describe the issue:

When I executed the code(See Reproduce the code example) and the segmentation fault was occurred.
I wrote the traceback message by GDB in Error message field.

if "device = tlf.CreateFirstDevice()" is commented out, the segmentation fault is not occurred.

### Reproduce the code example:

```python
from pypylon import pylon
from pypylon import genicam

tlf = pylon.TlFactory.GetInstance()
device = tlf.CreateFirstDevice()
```

### Error message:

```shell
Starting program: /home/ccsa/miniconda3/envs/240328_ex1/bin/python test.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe15db700 (LWP 2868)]
[New Thread 0x7fffe0dda700 (LWP 2869)]
[New Thread 0x7fffdab1e700 (LWP 2870)]
[Thread 0x7fffdab1e700 (LWP 2870) exited]
[Thread 0x7fffe0dda700 (LWP 2869) exited]
[Thread 0x7fffe15db700 (LWP 2868) exited]

Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fffe2340134 in ?? ()
(gdb) bt
#0  0x00007fffe2340134 in ?? ()
#1  0x00007ffff57c0896 in (anonymous namespace)::run (p=<optimized out>)
    at /opt/conda/conda-bld/gcc-compiler_1654084175708/work/gcc/libstdc++-v3/libsupc++/atexit_thread.cc:80
#2  (anonymous namespace)::run () at /opt/conda/conda-bld/gcc-compiler_1654084175708/work/gcc/libstdc++-v3/libsupc++/atexit_thread.cc:105
#3  0x00007ffff7c888a7 in __run_exit_handlers (status=0, listp=0x7ffff7e2e718 <__exit_funcs>, run_list_atexit=run_list_atexit@entry=true, 
    run_dtors=run_dtors@entry=true) at exit.c:108
#4  0x00007ffff7c88a60 in __GI_exit (status=<optimized out>) at exit.c:139
#5  0x00007ffff7c6608a in __libc_start_main (main=0x5e3fd0 <main>, argc=2, argv=0x7fffffffdea8, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffde98) at ../csu/libc-start.c:342
#6  0x00000000005e3ece in _start () at /usr/local/src/conda/python-3.12.2/Parser/parser.c:35620
```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

 CPU architecture:  X86_64
 Operating System: Ubuntu 20.04.6
   Miniconda3(conda 24.1.2)  python=3.12.2  pypylon=3.0.1
   Pylon 7.4
 RAM : 4GB
 Camera: acA1440-73gm
 No switches or hubs.
 cable: Ethernet Cable (Part No. 200027038)
  

### Runtime information:

```shell
python: 3.12.2 | packaged by Anaconda, Inc. | (main, Feb 27 2024, 17:35:02) [GCC 11.2.0]
platform: linux/x86_64/5.15.0-100-generic
pypylon: 3.0.1 / 7.4.0.38864
```

thiesmoeller, 2024-04-01:
Did you compile pypylon on your own ? 
As we have Python 3.12 Support currently only in a branch ?

kkuranoCCS, 2024-04-02:
Yes, but I used the default branch.
So, I have compiled pypylon using the branch "build-for-python3.12".
However,  segmentation fault has been occurred again.

Current runtime information:
python: 3.12.2 | packaged by Anaconda, Inc. | (main, Feb 27 2024, 17:35:02) [GCC 11.2.0]
platform: linux/x86_64/5.15.0-100-generic
pypylon: 3.0.1.dev4 / 7.4.0.38864

GDB message:
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fffe2340134 in ?? ()
(gdb) bt
#0  0x00007fffe2340134 in ?? ()
#1  0x00007ffff57c0896 in (anonymous namespace)::run (p=<optimized out>)
    at /opt/conda/conda-bld/gcc-compiler_1654084175708/work/gcc/libstdc++-v3/libsupc++/atexit_thread.cc:80
#2  (anonymous namespace)::run ()
    at /opt/conda/conda-bld/gcc-compiler_1654084175708/work/gcc/libstdc++-v3/libsupc++/atexit_thread.cc:105
#3  0x00007ffff7c888a7 in __run_exit_handlers (status=0, 
    listp=0x7ffff7e2e718 <__exit_funcs>, 
    run_list_atexit=run_list_atexit@entry=true, run_dtors=run_dtors@entry=true)
    at exit.c:108
#4  0x00007ffff7c88a60 in __GI_exit (status=<optimized out>) at exit.c:139
#5  0x00007ffff7c6608a in __libc_start_main (main=0x5e3fd0 <main>, argc=2, 
    argv=0x7fffffffde88, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffde78)
    at ../csu/libc-start.c:342
#6  0x00000000005e3ece in _start ()
    at /usr/local/src/conda/python-3.12.2/Parser/parser.c:35620

thiesmoeller, 2024-04-05:
can you check if the error persists, if you disconnect all "real" cameras and call your test as:

`PYLON_CAMEMU=1 /home/ccsa/miniconda3/envs/240328_ex1/bin/python test.py`

kkuranoCCS, 2024-04-08:
the following messages has been displayed.

(240328_ex1) ccsa@ubuntu:~/basler$ PYLON_CAMEMU=1
(240328_ex1) ccsa@ubuntu:~/basler$ /home/ccsa/miniconda3/envs/240328_ex1/bin/python test.py
Traceback (most recent call last):
  File "/home/ccsa/basler/test.py", line 7, in <module>
    device = tlf.CreateFirstDevice()
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ccsa/miniconda3/envs/240328_ex1/lib/python3.12/site-packages/pypylon/pylon.py", line 1720, in CreateFirstDevice
    return _pylon.TlFactory_CreateFirstDevice(self, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
_genicam.RuntimeException: No device is available or no device contains the provided device info properties. : RuntimeException thrown (file 'TlFactory.cpp', line 694)
Segmentation fault (core dumped)
(240328_ex1) ccsa@ubuntu:~/basler$ echo $PYLON_CAMEMU
1
(240328_ex1) ccsa@ubuntu:~/basler$

kkuranoCCS, 2024-05-21:
We have been had the same problem without Conda.
Environment:
```
Ubuntu 20.04.6
Python 3.8.10
pypylon 3.0.1
Pylon 7.4.0.14900
No Camera (Using PYLON_CAMEMU=1)
```

The python code is as follows. When using event registration, segmentation fault has been occurred.
```
from pypylon import genicam
from pypylon import pylon

UserProvidedId = 100

class SampleCameraEventHandler(pylon.CameraEventHandler):
    def OnCameraEvent(self, camera, userProvidedId, node):
        if userProvidedId == UserProvidedId:
            print("event")

handler = SampleCameraEventHandler()
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
camera.GrabCameraEvents.Value = True
camera.RegisterCameraEventHandler(handler, "ExposureEndEventData", UserProvidedId, pylon.RegistrationMode_ReplaceAll, pylon.Cleanup_None)
````

GDB back trace is as follows. It shows that the CInstantCamera destructor failed to release some kind of resources.
````
(gdb) bt
#0  0x0000000000ed3bd0 in ?? ()
#1  0x00007ffff69d5ac6 in ?? () from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#2  0x00007ffff69d5b7c in ?? () from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#3  0x00007ffff68b4934 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
   from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#4  0x00007ffff69d17ab in ?? () from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#5  0x00007ffff69d18f7 in ?? () from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#6  0x00007ffff69dfe05 in ?? () from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#7  0x00007ffff69dffb7 in ?? () from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#8  0x00007ffff69ccb60 in Pylon::CInstantCamera::~CInstantCamera() ()
   from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#9  0x00007ffff69ccb73 in Pylon::CInstantCamera::~CInstantCamera() ()
   from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/libpylonbase.so.7.4
#10 0x00007ffff48d4973 in _wrap_delete_InstantCamera ()
   from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/_pylon.cpython-38-x86_64-linux-gnu.so
#11 0x00007ffff48a9db8 in SwigPyObject_dealloc ()
   from /home/ccsa/.local/lib/python3.8/site-packages/pypylon/_pylon.cpython-38-x86_64-linux-gnu.so
#12 0x00000000005b02c0 in ?? ()
#13 0x000000000058738d in ?? ()
#14 0x00000000005ad05b in PyDict_Clear ()
#15 0x00000000006a510a in ?? ()
#16 0x00000000004c518f in ?? ()
--Type <RET> for more, q to quit, c to continue without paging--
#17 0x000000000066721f in _PyGC_CollectNoFail ()
#18 0x000000000067a634 in PyImport_Cleanup ()
#19 0x000000000067423f in Py_FinalizeEx ()
#20 0x00000000006b418d in Py_RunMain ()
#21 0x00000000006b43fd in Py_BytesMain ()
#22 0x00007ffff7de3083 in __libc_start_main (main=0x4c4510 <main>, argc=2, argv=0x7fffffffdf88, init=<optimized out>, 
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdf78) at ../csu/libc-start.c:308
#23 0x00000000005da67e in _start ()
````

thiesmoeller, 2024-05-21:
@kkuranoCCS 
It seems that the event handlers and the instant camera are not released in the proper order when your program exits.

Can you unregister the event handler before exiting your code

kkuranoCCS, 2024-05-22:
@thiesmoeller 
Thank you. I could avoid segmentation fault adding following code before exiting.

`camera.DeregisterCameraEventHandler(handler,"ExposureEndEventData")`