basler/pypylon #742: Obtaining distances for each pixel in the depth image

astef3, 2024-04-16:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

HI, 

first I would like to say that you did an outstanding job with the library. 
Currently what I am trying to do is to obtain image with Basler Blaze 101 using Pypylon library and filter out all the pixels that are higher or lower from a specific distance. So in a way I want to perform thresholding based on distance. However in order to achieve that I need the real metric distances. But when I used the opencv sample scirpt or similar examples in you samples code I obtain the grayscale image where the intensities of the pixels are based on the depth, but I would prefer if I could have the metric information instead. 

I will be extremely grateful for any help for this problem. I can provide more data if needed. 

Best regards, 
Aleksandar

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Camera: Basler Blaze 101

### Runtime information:

```shell
/
```

cpt-wojtech, 2024-04-22:
Here is a description with some C++ code of how to calculate 3D coordinates from the 2D Depth Map

https://docs.baslerweb.com/processing-measurement-results#calculating-3d-coordinates-from-the-2d-depth-map

There is nothing for Python but maybe the C++ can guide you in the right direction. 
By the way it is also possible to set a confidence threshold in the camera to filter out certain distances

https://docs.baslerweb.com/confidence-threshold

thiesmoeller, 2024-07-03:
We are close to releasing full support for blaze in pypylon
If you wan't to test the last RC:

```shell
pip install pypylon==4.0.0rc2
```