basler/pypylon #744: Distortion correction for Blaze 101 / 102

andangthanh, 2024-04-18:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi,

I'm using a blaze ToF camera and pypylon to process pointclouds, but the resulting pointclouds seem to distorted.
Does the camera internally do distortion correction before calculating the depth map as it does with the blaze viewer automatically?

If so, is it also possible to turn it off to show the difference if activated or to calculate a custom distortion correction?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Nvidia Jetson

### Runtime information:

```shell
python 3.9
```

SMA2016a, 2024-04-20:
For any questions, feel free to check out the Basler document page or get in touch with Basler’s local support or distributor, if the question is not directly points to pypylon.

https://docs.baslerweb.com/distortion-correction

PKowal181, 2024-04-25:
@SMA2016a 
"
Not Found
The requested URL was not found on this server.

Additionally, a 404 Not Found error was encountered while trying to use an ErrorDocument to handle the request.
"

SMA2016a, 2024-04-25:
when I have posted, there was a site. Check with your local distributor

astraloby, 2024-04-27:
Hello I have the same question. I will be using blaser 2 cameras for 3D motion tracking (crack formation on a concrete sample) and someone is helping me with the code and stuff. I don't know if the images retrieved are distortion-corrected or not. I have some notion of that distorted images are corrected with the cam parameters and that can be done after capturing the pics. If you have any information about it i would greatly appreciate that, thank you in advance.

SMA2016a, 2024-04-29:
@astraloby @PKowal181 
the link that I have posed working now.