basler/pypylon #747: How to utilize buffer when recording.

da9da9, 2024-04-22:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hi 
Thank you for providing pypylon.

I want to capture image by using pypylon, and my Basler cameras, =acA720-520um= and =Basler MED ace 5.1 MP 75 color=

The sequential capture strategy is that I want to record high speed at early stage of my experiment for several tens of seconds, and move on to the slow speed capture during the live view. So I imported cv2.
For example, you can find 
    total_frames = [1000, 1000]  # Frames at different stages [first stage, second stage]
    frame_rates = [400, 20]  # Corresponding frame rates [first stage, second stage]
from my script below. 
If this strategy is possible with pylon viewer software, It would be nice, but I could not find the feature, so I am trying to make a script.

I tried to set the buffer for the high-speed recording, but it did not use a buffer. It

I checked that 400 fps is possible with acA720-520um and pylon viewer, so I assume it is possible with python.

I will appreciate if you help me.

Thank you

```
import cv2
from pypylon import pylon
import time
import os
import numpy as np

def main():
    # Set the directory where images will be saved
    save_directory = input("Enter the directory to save images: ")
    if not os.path.exists(save_directory):
        os.makedirs(save_directory)
        
    # Request a user comment to include in the filename
    user_comment = "FixedCommentForAllImages"  # Example fixed comment
    
    # Create an instant camera object with the camera device found first.
    camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())
    print("Using Camera: ", camera.GetDeviceInfo().GetModelName())

    # Open the camera to access settings.
    camera.Open()
    
    # Load the specified UserSet settings.
    camera.UserSetSelector = "UserSet1"
    camera.UserSetLoad.Execute()
    
    # Stop grabbing to change buffer settings
    if camera.IsGrabbing():
        camera.StopGrabbing()

    # Set the buffer size
    try:
        camera.MaxNumBuffer = 10000
        print("Buffer size set successfully.")
    except Exception as e:
        print(f"Failed to set buffer size: {str(e)}")
    
    # Start grabbing images again after setting buffer size
    camera.StartGrabbing(pylon.GrabStrategy_LatestImages, pylon.GrabLoop_ProvidedByUser)
    
    # Create windows to display the images and information.
    cv2.namedWindow('Live View', cv2.WINDOW_NORMAL)
    cv2.resizeWindow('Live View', 612, 512)
    cv2.namedWindow('Info', cv2.WINDOW_AUTOSIZE)

    # Define a converter for converting grabbed images.
    converter = pylon.ImageFormatConverter()
    converter.OutputPixelFormat = pylon.PixelType_BGR8packed
    converter.OutputBitAlignment = pylon.OutputBitAlignment_MsbAligned

    recording = False
    frame_count = 0
    total_frames = [1000, 1000]  # Frames at different stages [first stage, second stage]
    frame_rates = [400, 20]  # Corresponding frame rates [first stage, second stage]
    current_stage = 0
    messages = []  # List to store session messages

    while camera.IsGrabbing():
        grabResult = camera.RetrieveResult(5000, pylon.TimeoutHandling_ThrowException)
        
        if grabResult.GrabSucceeded():
            image = converter.Convert(grabResult).GetArray()
            cv2.imshow('Live View', image)

            # Calculate current buffer usage as a percentage
            buffer_usage = ((frame_count % camera.MaxNumBuffer.GetValue()) / camera.MaxNumBuffer.GetValue()) * 100

            if recording:
                filename = f"img_{frame_count:04d}_fr{frame_rates[current_stage]}_{user_comment}.tiff"
                cv2.imwrite(os.path.join(save_directory, filename), image)
                frame_count += 1

                if frame_count >= sum(total_frames[:current_stage + 1]):
                    if current_stage < len(total_frames) - 1:
                        current_stage += 1
                        messages.append(f"Switching to next recording stage at {frame_rates[current_stage]} FPS.")
                    else:
                        recording = False
                        messages.append("Recording completed.")
                        break

                time.sleep(1 / frame_rates[current_stage])  # Control the frame rate

            update_info_window(current_stage, frame_rates, total_frames, recording, camera.MaxNumBuffer.GetValue(), buffer_usage, messages)

            k = cv2.waitKey(1)
            if k == 27:  # ESC key to exit
                break
            elif k == ord('r'):  # 'r' key to start recording
                if not recording:
                    recording = True
                    frame_count = 0
                    current_stage = 0
                    messages.append("Recording started.")
            elif k == ord('t'):  # 't' key to stop recording
                if recording:
                    recording = False
                    messages.append("Recording stopped.")

        grabResult.Release()
    
    # Stop grabbing and close camera and windows.
    camera.StopGrabbing()
    camera.Close()
    cv2.destroyAllWindows()
    #%%
def update_info_window(current_stage, frame_rates, total_frames, recording, buffer_size, buffer_usage, messages):
    info_text = f"Buffer Size: {buffer_size}\nBuffer Usage: {buffer_usage:.2f}%\nRecording: {'Yes' if recording else 'No'}\n"
    info_text += "Stages:\n"
    for i in range(len(frame_rates)):
        info_text += f"Stage {i+1}: {total_frames[i]} frames at {frame_rates[i]} FPS\n"
    info_text += "Messages:\n" + "\n".join(messages)

    info_image = 255 * np.ones(shape=[500, 600, 3], dtype=np.uint8)
    text_lines = info_text.split('\n')
    for i, line in enumerate(text_lines):
        cv2.putText(info_image, line, (10, 20 + i * 20), cv2.FONT_HERSHEY_SIMPLEX, 0.5, (0, 0, 0), 1)
    cv2.imshow('Info', info_image)
    
#%%

if __name__ == '__main__':
    main()

```

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

acA720-520um and Basler MED ace 5.1 MP 75 color
USB3 communication with PC.

### Runtime information:

```shell
OS is windows 11 

pylon Viewer 64-Bit
pylon Release 7.4.0.14900
pylon C++ SDK 7.4.0.38864
pylon Application 2.1.0.14100
pylon Viewer 7.4.0.14100

python: 3.11.7 | packaged by Anaconda, Inc. | (main, Dec 15 2023, 18:05:47) [MSC v.1916 64 bit (AMD64)]
platform: win32/AMD64/10
pypylon: 3.0.1 / 7.4.0.38864
```

SMA2016a, 2024-04-24:
since you are using enough buffer, I would try onenyone GrabStrategy  or set the camera.OutputQueueSize.Value = 10000 too.

if you want to rung the camera at 20fps mode you can use the parameter camera.AcquisitionFrameRate to define your target fps

da9da9, 2024-04-25:
THank you SMA2016. Your comment gave me a huge hint to resolve the issue

But another problem occurred. I will post another one. please help me.

closed 2024-04-25 by da9da9