basler/pypylon #749: Error: 3774873620 The buffer was incompletely grabbed.Please write a comprehensive title

DORSA-co, 2024-04-28:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

hi
i user ubuntu20 and pypylon.
i have 24 basler 1920 40gc caemra and 10g network.
when i run and getpicture i see this error :
Error: 3774873620 The buffer was incompletely grabbed.
how can i fix this ?

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Ram : 128 GB
cpu : AMD Ryzen Threadripper PRO 3975WX 
operating : ubuntu 20
camera : 24 * 1920 40gm

### Runtime information:

```shell
pylon info: 5.1.0.12682 64-Bit
----------------------------------------
pylon Viewer info: 5.1.0.12682 64-Bit
----------------------------------------
Host system infos:

CPU architecture: x86_64
Kernel type: linux
Kernel version: 5.15.0-105-generic
OS type: ubuntu
OS version: 20.04
----------------------------------------
Host network infos:

Host network adapter enp9s0:
    Name: enp9s0
    MAC address: 18:C0:4D:DB:40:C9
    IP address: 192.168.1.130
    Subnet mask: 255.255.255.0

Host network adapter virbr0:
    Name: virbr0
    MAC address: 52:54:00:64:02:C5
    IP address: 192.168.122.1
    Subnet mask: 255.255.255.0

Host network adapter virbr0-nic:
    Name: virbr0-nic
    MAC address: 52:54:00:64:02:C5

Host network adapter docker0:
    Name: docker0
    MAC address: 02:42:38:2D:E9:5D
    IP address: 172.17.0.1
    Subnet mask: 255.255.0.0
----------------------------------------
Device infos:

GigE Vision device:
    Model name: acA1920-40gc
    Manufacturer name: Basler
    Serial number: 24669046
    MAC address: 00:30:53:47:3E:76
    IP address: 192.168.1.101
    Subnet mask: 255.255.255.0

GigE Vision device:
    Model name: acA1920-40gm
    Manufacturer name: Basler
    Serial number: 23683746
    MAC address: 00:30:53:38:35:A2
    IP address: 192.168.1.100
    Subnet mask: 255.255.255.0
```

SMA2016a, 2024-04-29:
running all camera in freerun mode?
is PTP enabled?
did you optimize your system as written in pylon README? 
what is the pixelformat of the camera 
what is the target fps?

DORSA-co, 2024-04-29:
> running all camera in freerun mode? is PTP enabled? did you optimize your system as written in pylon README? what is the pixelformat of the camera what is the target fps? did you increase the packet size and interpacket delay as per the error message suggesting?

hi all camera is on trigger mode Line1.
no i dont enables PTP.
target fps is 8 fps with hardware trigger.
yes my packet size is 8000 and i increase the interpacketdelay.

What do you mean by this question  :  optimize your system as written in pylon README?

my transport layer in database is - packetsize : 8000 , interpacketdelay : 26000 , frame transmission delay : 5 group from 0 step 10000 , bandwidth reserve : 40 , bandwidth aqumulation : 5 .

and also i use opcua in my network for connect to plc siemens s7-1200.
is it possible my problem caused by that opcua connection ?

SMA2016a, 2024-04-29:
bandwidth reserve : 40   set it to 5.

<What do you mean by this question : optimize your system as written in pylon README?> Thread priority 

refer this document:
https://docs.baslerweb.com/network-configuration-(gige-cameras)

ReyhaneAbtahi, 2024-05-07:
> bandwidth reserve : 40 set it to 5.
> 
> <What do you mean by this question : optimize your system as written in pylon README?> Thread priority
> 
> refer this document: https://docs.baslerweb.com/network-configuration-(gige-cameras)

> @kesaroid did you find any solution or you still need help to setup your system?
Hello,

After numerous efforts, we realized that continuously measuring the GPU temperature using the GPUtil module was leading to this issue. The code we used for this task is as follows:

```python
gpus = GPUtil.getGPUs()
gpu_temp = gpus[0].temperature
```

After removing this line of code, our error rate significantly reduced.
This occurrence is very strange for us because the above code does not create any network connections and does not add any load on the network. Can you help us figure out why this is happening?

thiesmoeller, 2024-05-07:
Switch to use https://pypi.org/project/nvidia-ml-py/

Calling nvidia-smi is querying a lot more than you need

SMA2016a, 2024-12-26:
This topic was closed 60 days after the last reply. New replies are no longer wished on this thread.

closed 2024-12-26 by SMA2016a