basler/pypylon #750: fail when install pypylon on raspberry pi

lytandat, 2024-05-07:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello, 
I am trying to install the pypylon library onto the raspberry pi, I have followed the steps as instructed but in the last step pip3 install the installation path has an error, my error is as shown in the image below, do you have a solution? Can anyone fix this error so I can install the library on my raspberry pi?
[image]

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

ac2500

### Runtime information:

```shell
i'm using python 3.9.2
```

lytandat, 2024-05-07:
my RPi runs on ARMV7L

thiesmoeller, 2024-05-07:
Have you tried installing with 

pip install pypylon

lytandat, 2024-05-07:
thank you so much

closed 2024-05-08 by thiesmoeller