basler/pypylon #754: I want to read temperatures at different parts of the camera but it is not possible

simonmichel1996, 2024-05-23:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I want to read temperatures at different parts of the camera at runtime, but it looks like the temperature can only be read from the default spot (which is 'Coreboard'). If I change the value of cap.DeviceTemperatureSelector it will crash.

Here is some example code of the working configuration:

```
    tlf = pylon.TlFactory.GetInstance()
    tl = _lf.CreateTl("BaslerUsb")
    dev = tl.CreateFirstDevice()
    cap = pylon.InstantCamera()
    cap.Attach(dev)
    cap.Open()
    cap.StartGrabbing()
    # cap.DeviceTemperatureSelector.Value = 'Sensor'
    print(_cap.DeviceTemperatureSelector.Value)
    while True:
        print(cap.DeviceTemperature.Value)
```

Output:
```
Coreboard
44.81200000000001
44.81200000000001
...
```

If I uncomment the line `cap.DeviceTemperatureSelector.Value = 'Sensor'` however I get the following Output:

```
Sensor
Traceback (most recent call last):
  File "basler_handler.py", line 429, in <module>
    print(cap.DeviceTemperature.Value)
  File "/venv/lib/python3.10/site-packages/pypylon/genicam.py", line 1834, in GetValue
    return _genicam.IFloat_GetValue(self, Verify, IgnoreCache)
_genicam.AccessException: Node is not readable. : AccessException thrown in node 'DeviceTemperature' while calling 'DeviceTemperature.GetValue()' (file 'FloatT.h', line 293)
```

I got it running by using `cap.StopGrabbing()` before reading the temperature and `cap.StartGrabbing()` afterwards, but this does horrible things to the framerate.

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

Camera: [a2A5328-15ucBAS](https://docs.baslerweb.com/a2a5328-15ucbas.html)
CPU: Apple M2 Pro
OS: macOS Ventura (13.5)
RAM: 32GB

Interface: USB 3.1 (I don't know the specifics of the interface card)
Hubs: Yes, USB-A 3.1 to USB-C 3.1 Hub
Cable: 1m USB 3.1 Cable

### Runtime information:

```shell
python: 3.10.4 (main, Oct 26 2023, 09:55:43) [Clang 14.0.3 (clang-1403.0.22.14.1)]
platform: darwin/arm64/22.6.0
pypylon: 3.0.1 / 7.3.1.9
```

SMA2016a, 2024-05-24:
would you check with pylon viewer if the node DeviceTemperatureSelector has got the entry sensor?

simonmichel1996, 2024-05-24:
Yes, the entry is there in the pylon viewer and it's selectable

HighImp, 2024-06-07:
Its a little bit tricky: The Sensor State is "Standby" after wakeup, but must be "On" to read out the temperature.
During "StartGrabbing", the Sensor switch implicit to "On", but during acquisition, the temperature value of the image-sensor is not readable. So you have to stop the grabbing before sensor temperature readout. 
See also: https://docs.baslerweb.com/sensor-state.html