basler/pypylon #756: Camera Connection İssue

Tahir-Vural-Inovako, 2024-05-23:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

Hello,

We are using 8 Basler cameras within the scope of a project. All cameras grab images on separate threads. And FPS per camera is 1.25. There are 4 USB 3.0 ports on the computer with 2 PCI Express cards installed. There are a total of 8 USB 3.0 ports, and the cameras are connected to these ports. The cables from the cameras to the USB ports vary in length, ranging from 3 to 8 meters. We frequently experience connection losses with 1 or 2 of our cameras. The cameras that lose connection can have either short or long cables. Therefore, we don't think the connection loss is due to the cable length. Sometimes we experience the camera losing connection while the system and cameras are not working. Additionally, we sometimes receive an error suddenly while the system and cameras are working.

Some of the error messages we often receive are as follows:

Failed to retrieve XML file from camera device 'xxxx:xxxx:x:x:xx. Error: 'The requested operation timed out.'
Write operation failed on device 'xxxx:xxxx:xx:x:x' at address: 0xxxxxxxxx; Error: The requested operation timed out.
We previously sought support from Basler regarding this issue. However, they directed us to the manufacturer, and we didn't get any results. Can you provide support on this matter through this platform?"

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

GPU: RTX 4070
CPU : i7 12700
Ubuntu Linux 2204
RAM: 32 GB

### Runtime information:

```shell
Containerized applications are not used
```

SMA2016a, 2024-05-24:
this issue is not related to pypylon. We are support only pypylon related questions in forum. Please be so kindly turn to your loacal support team.

Tahir-Vural-Inovako, 2024-05-24:
Hi,
As seen in the picture, I am getting a USB 2.0 port error. However, I am sure that my USB cable and USB input port are 3.0. Why am I getting this error? I would be happy if you help.

[image]

cpt-wojtech, 2024-05-24:
Hello Tahir-Vural-Inovako,
you can contact your local Basler support team in order to handle this camera related issue 
https://www.baslerweb.com/en/support/contact/

Tahir-Vural-Inovako, 2024-06-06:
Hi,
We requested firmware for our Basler cameras. We also contacted the local basler support team. We explained the problems we experienced in detail. We also provided the required hardware details. We need this firmware very urgently. We'd love it if you could help.

SMA2016a, 2024-06-18:
what USB Card are you using? Why do you think that the firmware update will fix the issue? It sounds to me unstable power on USB BUS. what action is needed to get back the camera working?