basler/pypylon #759: Application Settings for High Performance

dahleri, 2024-06-14:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I have an application with real-time requirements, where I use the grab loop provided by the camera. I have problems with jitter, grab errors and skipped images when running the application, but not in the Pylon viewer. Jitter is identified by measuring the number of available buffers.

My grab configuration:
- Two cameras synchronised with PTP free run
- Grab loop provided by the camera
- GrabStrategy LatestImages

How can I improve the performance in my application to avoid said problems and make sure I always 

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

- Windows 11, 
- CPU-architecture x64 with 16 cores
- Cameras connected through a switch 
- GigE cameras 

### Runtime information:

```shell
Pylon v7.4
Windows 11
```

SMA2016a, 2024-06-17:
did you use GigE configurator to optimize your system? If this does not help, turn to your local support team, please.

dahleri, 2024-06-19:
Thank you for your reply! I have tried the GigE configurator and will contact my local support team instead.

closed 2024-06-22 by dahleri