basler/pypylon #760: BUG: Segfault in Pylon viewer

ethanc8, 2024-06-26:
### Describe the issue:

(this is not a bug on pypylon, it's a bug on pylon itself or in Qt)

Hi, I installed pylon using the [deb package](https://www2.baslerweb.com/media/downloads/software/pylon_software/pylon_7.3.0.27189_linux-x86_64_debs.tar.gz). When I run `/opt/pylon/bin/pylonviewer`, I get a segfault. The backtrace from gdb is:

```
(gdb) run
Starting program: /opt/pylon/bin/pylonviewer 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Downloading 0.12 MB separate debug info for /lib/x86_64-linux-gnu/libdrm.so.2
Downloading 0.07 MB separate debug info for /lib/x86_64-linux-gnu/libwayland-client.so.0                                                                                                                         
Downloading 0.11 MB separate debug info for /lib/x86_64-linux-gnu/libwayland-server.so.0                                                                                                                         
                                                                                                                                                                                                                 
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff59adf57 in wl_list_insert (list=0x555555684380, elm=elm@entry=0x5555556a59f0) at ../src/wayland-util.c:50
Download failed: Invalid argument.  Continuing without source file ./build/../src/wayland-util.c.
50      ../src/wayland-util.c: No such file or directory.
(gdb) bt
#0  0x00007ffff59adf57 in wl_list_insert (list=0x555555684380, elm=elm@entry=0x5555556a59f0) at ../src/wayland-util.c:50
#1  0x00007ffff59adfdb in wl_proxy_create_wrapper (proxy=0x5555556842a0) at ../src/wayland-client.c:2456
#2  0x00007ffff5ca226f in ?? () from /lib/x86_64-linux-gnu/libEGL_mesa.so.0
#3  0x00007ffff5c957e0 in ?? () from /lib/x86_64-linux-gnu/libEGL_mesa.so.0
#4  0x00007ffff5c8658f in ?? () from /lib/x86_64-linux-gnu/libEGL_mesa.so.0
#5  0x00007ffff7e8cf27 in ?? () from /opt/pylon/lib/Qt/plugins/wayland-graphics-integration-client/libqt-plugin-wayland-egl.so
#6  0x00007ffff186f59d in QtWaylandClient::QWaylandIntegration::initializeClientBufferIntegration() () from /opt/pylon/lib/Qt/plugins/platforms/../../lib/libQt5WaylandClient.so.5
#7  0x00007ffff1870115 in QtWaylandClient::QWaylandIntegration::clientBufferIntegration() const () from /opt/pylon/lib/Qt/plugins/platforms/../../lib/libQt5WaylandClient.so.5
#8  0x00007ffff186e87f in QtWaylandClient::QWaylandIntegration::createPlatformOpenGLContext(QOpenGLContext*) const () from /opt/pylon/lib/Qt/plugins/platforms/../../lib/libQt5WaylandClient.so.5
#9  0x00007ffff6d761c1 in QOpenGLContext::create() () from /opt/pylon/bin/../lib/Qt/lib/libQt5Gui.so.5
#10 0x00007ffff6d2d1a0 in QGuiApplicationPrivate::init() () from /opt/pylon/bin/../lib/Qt/lib/libQt5Gui.so.5
#11 0x00007ffff7571dad in QApplicationPrivate::init() () from /opt/pylon/bin/../lib/Qt/lib/libQt5Widgets.so.5
#12 0x000055555555fc72 in ?? ()
#13 0x000055555555accb in ?? ()
#14 0x00007ffff5a29d90 in __libc_start_call_main (main=main@entry=0x55555555aab0, argc=argc@entry=1, argv=argv@entry=0x7fffffffd4e8) at ../sysdeps/nptl/libc_start_call_main.h:58
#15 0x00007ffff5a29e40 in __libc_start_main_impl (main=0x55555555aab0, argc=1, argv=0x7fffffffd4e8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffd4d8)
    at ../csu/libc-start.c:392
#16 0x000055555555bfbe in _start ()
```

### Reproduce the code example:

```python
# Not applicable
```

### Error message:

```shell
Segmentation fault
```

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup & camera model(s) used

x86_64 Intel CPU, KDE neon (based on Ubuntu 22.04; Qt is built by KDE and not Canonical), 7.5 GiB RAM

Camera is a Basler acA4024-26um

### Runtime information:

```shell
python: 3.11.9 | packaged by conda-forge | (main, Apr 19 2024, 18:36:13) [GCC 12.3.0]
platform: linux/x86_64/6.5.0-26-generic
pypylon: 3.0.1 / 7.4.0.38864
```

ethanc8, 2024-06-26:
Note that pypylon continues to work.

ethanc8, 2024-06-26:
Forcing pylon viewer to use X11 works:
```bash
QT_QPA_PLATFORM=xcb /opt/pylon/bin/pylonviewer
```

bjoernrennfanz, 2024-07-11:
Please try the newest pylon update, 7.5 is out. This hast full support for wayland backends.

jminz, 2024-08-18:
> Forcing pylon viewer to use X11 works:
> 
> ```shell
> QT_QPA_PLATFORM=xcb /opt/pylon/bin/pylonviewer
> ```
QT_QPA_PLATFORM=xcb QT_DEBUG_PLUGINS=1 /opt/pylon/bin/pylonviewer
Yes, and don't forget to install libxcb-cursor0