basler/pypylon #761: BUG: Cannot set exposure time using acA4024-29um

ethanc8, 2024-06-26:
### Describe the issue:

Hello,

I can't set the exposure time of the acA4024-29um camera. Querying `genicam.IsAvailable(camera.ExposureTime))` returns `False` and actually attempting to set the exposure time results in an exception.

### Reproduce the code example:

```python
from pypylon import pylon, genicam

# conecting to the first available camera
camera = pylon.InstantCamera(pylon.TlFactory.GetInstance().CreateFirstDevice())

print(genicam.IsAvailable(camera.ExposureTime))

camera.ExposureTime.Value = 1000 #ms
```

### Error message:

```shell
False
Traceback (most recent call last):
  File "/home/ethan/Projects/SummerPrograms/Fermilab/BREAD/BREADVision/camview.py", line 17, in <module>
    camera.ExposureTime.Value = 1000 #ms
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ethan/miniforge3/envs/BREADVision/lib/python3.11/site-packages/pypylon/genicam.py", line 2091, in SetValue
    return _genicam.IFloat_SetValue(self, Value, Verify)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
_genicam.AccessException: Node is not writable. : AccessException thrown in node 'ExposureTime' while calling 'ExposureTime.SetValue()' (file 'FloatT.h', line 85)
```

### Is your camera operational in Basler pylon viewer on your platform

No

### Hardware setup & camera model(s) used

acA4024-29um

### Runtime information:

```shell
python: 3.11.9 | packaged by conda-forge | (main, Apr 19 2024, 18:36:13) [GCC 12.3.0]
platform: linux/x86_64/6.5.0-26-generic
pypylon: 3.0.1 / 7.4.0.38864
```

ethanc8, 2024-06-26:
Sorry, I didn't call `camera.Open()`

closed 2024-06-26 by ethanc8

SMA2016a, 2024-07-18:
I close the case now