basler/pypylon #762: BUG: Memory leakage when running in docker

Jannick-v, 2024-07-02:
### Describe the issue:

I am currently facing several issues in one of my projects. 
In an attempt to isolate the problem, I have tried to extract a MWE from the application code.
It is in this code that I have noticed a memory leak.

The real application runs in a `k3s` pod on a Jetson device.
In an attempt to isolate the issue, I am currently trying to step up from bare metal (conda env) --> docker env--> k3s env.
When running in a  docker container, the container crashes with an OOM.
the following docker run flags are used:

* `--network=host` : to make sure we can reach the camera
* `--memory=300m` : to limit the memory: the real application runs in k3s, where memory limits should be respected too.

I've tested this with a few docker base images, always with the same result:
- `python:3.8.16-slim` feature python 3.8
- `nvcr.io/nvidia/l4t-base:r36.2.0` featuring python 3.10

When running  inside the docker container, I can see a stead memory build-up happening, until the docker container eventually crashes. 
The camera (a `ral4096-24gm` Linescan camera)  is currently producing 800x420 mono images at a rate of ~ 4 FPS.(line rate ~3003.0 Hz)
The OOM happens after a few after starting the application.

### Reproduce the code example:

```python
import logging
import shutil
from pathlib import Path
import os
import time
from pypylon import pylon, genicam
from typing import List, Union, Optional
import gc
logger = logging.getLogger(__name__)

ROOT = Path(__file__).parent
IMG_DIR = Path(os.environ.get("IMG_DIR", ROOT / "images"))

PFS_FILE = ROOT / "camera.pfs"
SERIAL_NUMER = "24713383"
ZERO_COPY = True
CAMERA_IP = "10.102.3.53"
TIMEOUT = 5000

tl_factory = pylon.TlFactory.GetInstance()

    
def connect_cam_on_serial(serial_number:str) -> pylon.InstantCamera:
    cam = None
    for dev_info in tl_factory.EnumerateDevices():
        if dev_info.GetSerialNumber() == serial_number:
            cam = pylon.InstantCamera(tl_factory.CreateDevice(dev_info))
            break
    if cam is None:
        raise IOError(f"no camera with serial {serial_number} in {[d.GetFriendlyName() for d in tl_factory.EnumerateDevices()]}")
    else:
        cam.Open()
        logger.info(f"camera opened with serial {serial_number}")
        return cam        
    

def connect_cam_on_ip(ip:str) -> pylon.InstantCamera:
    device_info = pylon.DeviceInfo()
    device_info = device_info.SetAddress(AddressValue=ip)
    device_info.SetDeviceClass("BaslerGigE")
    cam = pylon.InstantCamera(tl_factory.CreateFirstDevice(device_info))
    cam.Open()
    logger.info(f"camera opened with ip {ip}")
    return cam

[... 193 lines cut ...]
 Runtimes: io.containerd.runc.v2 nvidia runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 8b3b7ca2e5ce38e8f31a34f35b2b68ceb8470d89
 runc version: v1.1.12-0-g51d5e94
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
 Kernel Version: 5.10.104-tegra
 Operating System: Ubuntu 20.04.6 LTS
 OSType: linux
 Architecture: aarch64
 CPUs: 8
 Total Memory: 29.87GiB
 Name: ubuntu
 ID: a3c92f65-f37a-46c4-a94a-d2ab0b1c0e1f
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
```
```

thiesmoeller, 2024-07-03:
a) could you move to the upcoming pypylon 4.0 ( available as 4.0.0rc2 on pypi) so we can check with latest fixes
    we fixed part of the zerocopy handling 
b) can you run your application using heaptrack 
    ```
    apt-get install heaptrack heaptrack heaptrack_gui
    ```
    
    this will alow you to generate a fine grained trace of the main allocators that leak in your example

thiesmoeller, 2024-07-04:
You set `GevSCPD	0` this is the delay time between gigevision packets.
As your camera is capable to sent packets with the minimum [inter packet gap](https://en.wikipedia.org/wiki/Interpacket_gap) of 96ns your receiver NIC and Linux kernel stack might be overrun.

The typical procedure to set this value is described here: https://docs.baslerweb.com/knowledge/how-to-troubleshoot-lost-packets-or-frames-while-using-gige-cameras

The idea is to increase the space between network packets ( and thus decrease the pressure on your hardware and software stack ) up to the point the virtual framerate of your linescan camera decreases.

I see, that you set a very very large number of buffers in your application. In a gigevision use case, where there is always a small packet queue to the hardware and from there the data is copied to your application buffers.
So adding more buffers in a continous acquisition use case is only needed, if your image processing *sporadically* is slower than the incoming image rate and you have to be sure that there will always buffers ready to be filled by the hardware packet queue.

thiesmoeller, 2024-07-05:
For your target k3s environment:
Which CNI do you use and in which configuration?

Buffer management is explained here https://docs.baslerweb.com/pylonapi/cpp/pylon_programmingguide#the-default-grab-strategy-one-by-one

Jannick-v, 2024-07-05:
Thanks for the link! 
I hope you have succesfully received the file?
We have an open support ticket (ref 74135710, + I took the liberty to already include you in mail trace) with more details, but currently we have the following:

We have tried different flannel/multus versions and different cni spec versions but without success.
Eventually, the below network config +  `pypylon 1.9.0` + `GevSCPD 125000`   seems to be the most stable combo so far.

```
root@ubuntu:/home/mic-733ao# kubectl -n kube-system get network-attachment-definition macvlan-local-link -o yaml
apiVersion: k8s.cni.cncf.io/v1
kind: NetworkAttachmentDefinition
metadata:
  annotations:
    objectset.rio.cattle.io/applied: H4sIAAAAAAAA/4yRTYvbQAyG/4rR2XbtfJjY0EOgt0KPPfkiy5pk6rHGeFSHEPLfy9TZZVlI2OO8D3r0irkBTvY3z8F6gQaGQ8hJbE5CJrf+21JCCoOVHhr4xXrx83BURTqPLPqDjRWrcTKFkRV7VITmBijiFSMI8em7P0waWPPZ+pxQ1XGU22iF9Cn3F+E5Oy1DLLYNH8hSpslPK/33Y9//3/5aITgyNDAiLQ4lE9YM36/40nSYkKJi+NtxFq5BeYR7Cg47di9vPGM4QwO1MUXRmZq7/WZfmaIiQ1WPSFsy1bas8dCXvKt3UfqprfOELnNWBljZky5hYopNyIuxJ2jglrRA8va5LTRJC0W+y4sW0qQFvU68ho9NazxiUJ5X4FA2jxTu938BAAD//7Okz+grAgAA
    objectset.rio.cattle.io/id: ""
    objectset.rio.cattle.io/owner-gvk: k3s.cattle.io/v1, Kind=Addon
    objectset.rio.cattle.io/owner-name: macvlan-net-attachment
    objectset.rio.cattle.io/owner-namespace: kube-system
  creationTimestamp: "2024-07-02T16:48:41Z"
  generation: 7
  labels:
    objectset.rio.cattle.io/hash: 9ff00bf9eb5256f06cfc6daac3cf6319a8d1e494
  name: macvlan-local-link
  namespace: kube-system
  resourceVersion: "61489"
  uid: f4d81b2b-cf3b-4f47-846a-71661c6824de
spec:
  config: '{ "cniVersion": "0.4.0", "type": "macvlan", "master": "lan2", "mode": "bridge",
    "mtu": 9000, "ipam": { "type": "host-local", "ranges": [ [ { "subnet": "169.254.0.0/16",
    "rangeStart": "169.254.184.200", "rangeEnd": "169.254.184.220" } ] ] } }'
```

Because we are nearing the project deadline, I have little room to keep on changing and testing things. 
Nevertheless, I still would like to see how we could improve in the future, so I am all ears :)

thiesmoeller, 2024-07-08:
There is no significant memory leak in your trace. 
Only leaks are in python itself ( maybe false positive )

You allocate in your code 10000 Buffers of 420x800 pix Mono8 and end up with 3.4GB Memory.
[image]

I assume, that you run into this, when your docker/k3s env fails

Jannick-v, 2024-07-08:
haha ok, that's pretty embarrassing :sweat_smile: 
just did a test without the `set_buffer_sizes` and can confirm.

We are still looking into the networking part, because an `inter packet delay` of `1 ms` still seems pretty high..

thiesmoeller, 2024-07-09:
Using MacVlan as cni in your case should give you the right performance.

Important is to use a CNI that routes the packets as early as possible into the pod.

Take care to increase the Rx mem size as you did.

Explanation to why it was not working with packet delay 0 and chunks:

The image data was " naturally" throttled by the line rate of your sensor. The chunks are send from camera buffers at full speed at the end of the transfer and are too close back to back for your host

Jannick-v, 2024-07-09:
All right. Thanks for the explanation.
Very much appreciated.
I think we can close this issue, since it was a `False` alarm.

closed 2024-07-09 by Jannick-v