basler/pypylon #764: Pyyylon leaving open files (or pipes) after stopping the cameras

cbenitez81, 2024-07-11:
### Describe what you want to implement and what the issue & the steps to reproduce it are:

I'm working in a workflow that requires that cameras are turn on and off periodically. I'm receiving a message for both cases but after repeating the process several times, I get an error for linux about too many open files. I could increase the limits in the limits.conf but the issue still remains. Below there is a minimal version of my program:

```
import pypylon.pylon as pylon
import psutil

factory = pylon.TlFactory.GetInstance()
info = pylon.DeviceInfo()
info.SetPropertyValue("IpAddress", "192.168.0.1")
p = psutil.Process()
print("number of files and pipes", p.num_fds())

for k1 in range(10):
    camera_object = pylon.InstantCamera(factory.CreateDevice(info), pylon.Cleanup_Delete)
    camera_object.Open()
    camera_object.Close()
    camera_object.DestroyDevice()
    print("number of files and pipes", p.num_fds())
    del camera_object

```

Result:
```
122
126
31
136
...
171
```

I tried with pypylon 4 but I got the same problem.

Any help is welcome, thanks!!

### Is your camera operational in Basler pylon viewer on your platform

Yes

### Hardware setup & camera model(s) used

CPU: X86
Ubuntu 22.04
RAM: 32 GB
Camera: Basler a2A1920-51gcPRO

### Runtime information:

```shell
python: 3.10.14 (main, May  6 2024, 19:42:50) [GCC 11.2.0]
platform: linux/x86_64/6.5.0-41-generic
pypylon: 3.0.1 / 7.4.0.38864
```

thiesmoeller, 2024-07-11:
So you power down your camera ?

As the cleanup_delete action and the device destroy is typically not used ....

Normally you just close the device ... The garbage collector will take care to cleanup everything.

But I assume that you already tried this ?

Regarding the described leak we have to investigate

cbenitez81, 2024-07-12:
That is correct, I forced garbage collection, but the number increases, even if you just Open and close the camera without destroying the object, the problems remains

SMA2016a, 2024-07-17:
Thanks for reporting this effect.  we can also see the effect. We will check and get back to you soon.

SMA2016a, 2024-07-18:
a fix sent via PN

closed 2024-07-19 by SMA2016a

fkdlqj13245, 2024-08-29:
@SMA2016a  Is it fixed on pypylon 4.0.0?
Because we has same problem with same version.